Ultimaker/Cura #7484: Top/bottom pattern optomization

Wdfitzpat007, 2020-04-12:
Having the lines top/bottom pattern make a ton of short printing moves is super annoying. Especially when they make a clear pattern.

**Describe the solution you'd like**
If the slicer sees that it is making a ton of these short moves in an obvious pattern, it should replace these moves with a concentric pattern with a line width matching the area

**Describe alternatives you've considered**
Manually editing areas of the top surface, very manual labor intensive.

**Affected users and/or printers**
All

An example is attached.
[FuNoTake3mfFile.zip](https://github.com/Ultimaker/Cura/files/4466553/FuNoTake3mfFile.zip)

**Additional context**
(Add any other context or screenshots about the feature request here.)

rburema, 2020-04-12:
We're working on these (and similar) sort of issues, but it's a long term project. See [libArachne](https://github.com/Ultimaker/libArachne), which we'll eventually merge into CuraEngine.

harikattar, 2020-05-15:
In the meantime, are there any workarounds?  If a part is small enough I can rotate it slightly to change the angle, but that doesn't work on narrow curved surfaces.

Ghostkeeper, 2020-05-19:
You could place a cutting mesh ("Override settings for overlap") in those thin areas with the following per-object settings:
* 0 walls.
* 100% infill or bottom thickness of 9999999mm.
* Top/Bottom Pattern to concentric.