Ultimaker/Cura #7486: Cura Inserts Tool Change Before Start Gcode

theOehrly, 2020-04-12:
**Application version**
4.5.0

**Platform**
Windows 10 64bit

**Printer**
Custom, two extruders

**Reproduction steps**
1. Add start gcode in printer settings
2. Slice and save

**Actual results**
In the resulting Gcode, the tool change for starting the print is inserted as the very first command before the start gcode.
Problem: My start gcode changes tool a few times. I can only guess which tool I need to end with.

I'm using {material_bed_temperature_layer_0, 0} for setting the hotend temperature. As far as I know, there is nothing like {first_tool} to end my start gcode code with?

**Expected results**
The first tool change by cura should be inserted right before the very first print move. That way it does not matter which tool the machine and extruder gcodes end with.

rburema, 2020-04-12:
By first tool, do you happen to mean `{initial_extruder_nr}`? :-)

For a more extensive discussion, see [here](https://github.com/Ultimaker/Cura/issues/6383).

theOehrly, 2020-04-13:
Yes, that is what I was looking for. Thank you.
Are those variables documented somewhere? I can't find it. Maybe there's more interesting stuff available that I just don't know of? :)

But, concerning GCode inserted before the start gcode. There is still a T1 command before my start gcode. It's only that I now also can insert the right command a the end of the start gcode. 

I don't think there should be any commands before the start gcode ever. I certainly didn't expect it. 
I know I can't and shouldn't rely on a specific tool being active. That's not my point. But it's the start gcode after all. I don't like the idea of surprise commands before my start gcode.

Maybe there is a very good reason which I don't see currently? I just wanted to give my opinion on that.

Ghostkeeper, 2020-04-21:
Those variables are documented in the Settings Guide: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/machine_settings/machine_start_gcode.md . This is also a plug-in you can find in the Marketplace.

There are three commands before the start g-code: The initial extruder, the printing temperature and the build plate temperature. Those are put there to make it easier to write start g-codes. The temperatures are omitted if the start g-code uses any temperature setting, because it assumes that you're setting it yourself then. However there is no way to prevent the initial extruder from being set. That's being requested in this ticket: https://github.com/Ultimaker/Cura/issues/7395

smoki3, 2020-04-21:
I have exactly the same issue on my tool changer

nehalem-make, 2021-08-06:
Same issue here when trying to generate GCODE for a Raise3D Pro2. I already have `T{initial_extruder_nr}` where it belongs, but Cura places a duplicate at the start of the GCODE anyway.  Some errata in Raise3D's proprietary firmware causes both extruders to heat unless the first tool change instruction is removed.

Ghostkeeper, 2021-08-17:
Closing this ticket as a duplicate of https://github.com/Ultimaker/Cura/issues/7395.

closed 2021-08-17 by Ghostkeeper