Ultimaker/Cura #7487: Disconnect supports

larsch, 2020-04-12:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
CFDM

**Reproduction steps**
Slice with support

**Screenshot(s)**
[image]

**Actual results**
Support towers not connected to bridges

**Expected results**
Supports towers connected to bridges

**Project file**
Attached

[CFFFP_5vPsuMount_PointlessSupport.zip](https://github.com/Ultimaker/Cura/files/4467169/CFFFP_5vPsuMount_PointlessSupport.zip)

Vandrasc, 2020-04-14:
Hi @larsch, in this case, you could disable the Use towers setting, in your model the support is needed only in the holes part if I see correctly.

Ghostkeeper, 2020-04-23:
I'm 99% sure we've had this report before a couple of times, but I can't find the previous discussions.

sgtnoodle, 2020-04-24:
I just installed 4.6 a couple days ago and noticed some new weirdness in supports.

Similar useless towers:
[image]
[image]
[image]

And this just seems wonky:
[image]

fvrmr, 2022-03-24:
We have added this ticket on our backlog.
Devs see CURA-9074