Ultimaker/Cura #7488: Custom layer height in per model settings

MrEd88, 2020-04-13:
**Is your feature request related to a problem? Please describe.**
Adaptative layer height is a good option, however it is difficult to get the best settings and avoid using higher or lower layers where we really want. 

**Describe the solution you'd like**
As stated [in reddit](https://www.reddit.com/r/ultimaker/comments/d4c26m/different_not_adaptative_layer_height/) "The main reason why this hasn't been implemented yet is because there needs to be some interface made for it, I'd say. That's probably 80% of the work here.", one simple solution to avoid adding a new interface, would be to allow changing layer heights in per model settings, using overlap.

**Describe alternatives you've considered**
Only option is to use adaptative layers, but they are not very useful to print something like minis bases or any other piece that requires to use different layer heights in especific areas.

**Affected users and/or printers**
All the people, all the printers. Mainly people printing minis or anything with bases. 

**Additional context**
This could be an option for #6840

fieldOfView, 2020-04-13:
CuraEngine can only handle a single arbitrary layer height per layer (with the exception where multiples of the lineheight are used, so lines can be printed eg every other layer). Using a per-model-setting to change layer heights would suggest you can change the layer height for any overlap between two models, which would not be possible.

MrEd88, 2020-04-13:
> CuraEngine can only handle a single arbitrary layer height per layer (with the exception where multiples of the lineheight are used, so lines can be printed eg every other layer). Using a [...]

Oh, I get it. I am used to custom supports and blockers using boxes, they intercept the model in a single upper and lower layer. 
The main idea/suggestion is to use the same method of adding boxes to define the heights with our desired layer heights, so if you use a different shape than a box the setting could be ignored. That may be a new category of per model settings only for layer change, so it can be easier to keep the settings apart. Make sense ? Too complicated?

MrEd88, 2020-04-21:
> I'm not sure if I'm visualising the same as you, but I'm interpreting you that you can only change the layer height if the modifier mesh overlaps with the entire layer(s), or at least within the printer's volume. However I think this is going to be very hard to discover, as it requires very specific positioning and resizing of meshes and also a very specific shape. If it gets too convoluted, for a user it would make more sense to integrate it in the layer slider or something.

Well, I have no idea of how complex would be to implement, but as far as I understand the bigger issue to custom layer heights is to define the intervals. So, I guess using a box (or even any other shape) without taking into consideration X and Y size and coordinates could be a good (simple?) way to implement it, we just need to know the Z intervals. Layer sliders the way PrusaSlicer do would be great, but I guess it is a lot of work.

Ghostkeeper, 2020-11-03:
I would be more in favour of having actual explicit per-layer settings rather than exposing the feature through the per-object settings.

truegret, 2021-11-28:
I respectfully request to re-open this request.  It is possible, @fieldOfView - My 4.1 version of Simplify3D does this easily.  Since you are print in actual 2.5D space, layer-by-layer (possible exception in Vase mode), you can inject a different layer height calculation starting at a given layer height.