Ultimaker/Cura #7490: Material reset to PLA

fieldOfView, 2020-04-13:
**Application version**
master (d0901c9), from source

**Platform**
Windows 10

**Printer**
UM S5

**Reproduction steps**
1. Start with a clear configuration
2. Add non-networked Ultimaker S5
3. Set configuration of second extruder to BB 0.4 and Generic PVA
4. Close Cura
5. Inspect `ultimaker_s5_extruder_right+%232.extruder.cfg`
-> 
```
...
[containers]
...
4 = generic_pva_ultimaker_s5_BB_0.4
```
As expected.

6. Restart Cura
-> Configuration of second extruder is changed to Generic PLA instead of Generic PLA, variant is still BB 0.4

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4468701/cura.log)

fieldOfView, 2020-04-13:
Likely culprit: https://github.com/Ultimaker/Cura/commit/caa348265dfad411053c267ce5060f2a518889fc

smartavionics, 2020-04-13:
> Configuration of second extruder is changed to Generic PLA instead of Generic PLA, variant is still BB 0.4

I think you mean "instead of Generic **PVA**"

commit 8269ae95 (2020-04-14): Use base_file to check if a material is known

Fixes #7490
commit 52f01a71 (2020-04-14): Use base_file to check if a material is known

Fixes #7490
closed 2020-04-14 by nallath