Ultimaker/Cura #7491: Resizing/scaling of model leads to corrupted slicing gcode

cyberrbob, 2020-04-13:
**Application version**
4.5.0

**Platform**
Windows 10.0.18363
Intel HD 530 / Nvidia Quadro M1000M (two GPU's)

**Printer**
Sovol SV01 ( similar to Creality CR-10)

**Reproduction steps**
1. Opened the [model](https://drive.google.com/file/d/1RS-smsuv1VGXzCiDTIxCCShZ-EBQAiym/view?usp=sharing) and resized/scaled it [image]

 from 53mm to 100mm it in Cura.
2. Pushed the 'slice' button in checked the result: [image]

**Actual results**
Please see step 2. This is not a gcode-visualization issue, problem exists in actual printed model (I saw this bug too late, only after print is done).

**Expected results**
No gaps in the sliced result are expected.

**Project file**
[here](https://drive.google.com/file/d/14UZ13t4KIX6z3QBya30ONr_VjILrwk7m/view?usp=sharing) it is.

**Log file**
[cura.LOG](https://github.com/Ultimaker/Cura/files/4469240/cura.LOG)

**Additional information**
If I resize the model in Microsoft 3D builder and slice it - there is no problem.

smartavionics, 2020-04-13:
The problem is that the model is not valid, it has errors....

[image]

The error regions are highlighted with that weird blue/purple pattern (that's a new cura feature that isn't in a release yet) but you can also see the warning shown by the cura mesh tools plugin (which is available now from the marketplace). Fix your model!

cyberrbob, 2020-04-13:
Hm, okay I see it now. Model is not mine (I have very limited knowledge of 3d modelling). But anyway, thanks for quick reply.

closed 2020-04-13 by cyberrbob