Ultimaker/Cura #7492: Cura 4.5 doesn't/can't open obj files (even the simplest ones)

alu000, 2020-04-13:
Hello. Like I said, I can't load any obj. files on cura, at least the ones made on C4D (or the ones from poser).

**Application version**
Cura 4.5

**Platform**
W7 virtualized via parallels on a MacBook Pro. Core 2 duo, 2.8 GHz, 8 GB ram (2 for W7).
Computer not connected to internet. Writing through the phone.

**Printer**
Ender 3 Pro.

**Reproduction steps**
1. Create a tetrahedron/cube on C4D and export it as obj file. Just that, no materials, no UV maps… no anything else: just create and export.
2. Trying to open it on cura.

**Actual results**
I get the message "Invalid file. Failed to load tetraedro prueba.obj"

**Expected results**
File loading/opening without problems.

**Project file**
So simple (8 lines) that I can write it in full. I exclude comments and this kind of things.
v 0 0 0
v 0 0 5
v 0 5 0
v 5 0 0
f 1 3 2
f 1 2 4
f 1 4 3
f 2 3 4

**Log file**
Wow, I never saw such a big .log file with so many warnings.
[cura.log](https://github.com/Ultimaker/Cura/files/4469256/cura.log)

Thanks in advance. Sergio.

smartavionics, 2020-04-13:
Please zip up and attach a sample obj file that cura fails to load, thanks.

smartavionics, 2020-04-13:
Actually, looking at the log file I can see that it can't find the file rather than failing when loading the file. Could you please try again but this time, don't have any spaces in the filename. Thanks.

smartavionics, 2020-04-13:
Interestingly, on my Linux machine I can load an obj file that has spaces in the file name so I'm not sure what's going wrong for you.

smartavionics, 2020-04-13:
The files it is failing to open seem to start with /Home/Documents (or similar) which isn't actually a Windows style path name so I wonder if the problem is that you are running a Windows Cura virtualized on a Mac and the file chooser is producing a mac style file name which is not being translated to a windows style file name?

alu000, 2020-04-13:
Hello Mark. You are 100000% right. I've copied the file on the virtualized W7 disk and… BINGO, the file loads smoothly (with or without spaces between words). I have my Mac home directory shared, and I see all the contents on windows explorer, but for any reason cura doesn't like, on windows, a Mac file structure (I guess the opposite will also be valid: my Mac cannot run osx el capitan or later, so I cannot check it). As all 3D objects I make for poser are on mac, and my home directory is shared, I thought that there wouldn't be any problems.
At least finally I'll be able to print with my ender, and of course things more complicated than a simple cube or tetrahedron.
Thanks a lot for your assistance and tips.Sergio.

smartavionics, 2020-04-13:
Hi Sergio, glad to hear that you can load obj files OK. I don't have any experience with Windows or Mac so I can't offer any suggestions as to making the file chooser produce a Windows style file name. Perhaps someone else can assist with that.

Ghostkeeper, 2020-04-21:
It makes sense that it won't read Windows style paths on a Mac version of Cura because we make heavy use of Python's `os.path.sep` and `os.path.join()` features, which look at the current operating system to navigate the file system. On Mac Cura `os.path.sep` would equate a forward slash and `os.path.join` joins paths using forward slashes, and then opening the resulting file with forward slashes would say that the file doesn't exist.

I'm not sure how we would approach fixing this. It's hard to guess which style path is necessary in this sort of situation.

GregValiant, 2026-05-30:
Is this still a problem in 5.12.1 or can this be closed?