Ultimaker/Cura #7493: Cura 4.5. Cube infill issues.

ranbros, 2020-04-13:
On cura 4.5 is supported anet a8. 
The standard infill is set to cubic but this produce artifact on surface due to infill that push the shell. 
With previous cura the default infill was lines. Lines works better. 
I have change the infill pattern from cubic to lines in Cura 4.5 this solves the problem. 

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

mahtDFR, 2020-04-14:
Thanks for the report. Please save a project file (File -> Save...), zip up the .3mf file and attach to this issue so we can try to reproduce it.

If you haven't already, please can you attach a [log file](https://github.com/Ultimaker/Cura#logging-issues).

ranbros, 2020-04-14:
[AA8_bird_whistle.zip](https://github.com/Ultimaker/Cura/files/4476112/AA8_bird_whistle.zip)

Ghostkeeper, 2020-04-21:
So I think you just need to reduce the [Infill Overlap](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/infill/infill_overlap.md) setting. It'll probably result in slightly weaker parts, but it'll look better.

I'm closing this since it's really working as expected. There is no difference between cubic and lines with the overlap. There is a difference due to the new Infill Overlap setting.

closed 2020-04-21 by Ghostkeeper