Ultimaker/Cura #7494: [4.2.1] sometimes generates bad gcode ending

klillas, 2020-04-13:
**Application version**
4.2.1

**Platform**
Windows 10

**Printer**
Creality CR-104S

**Reproduction steps**
Slice and store to SD card. This seems to apply to any STL randomly. Sometimes it generates gcode correctly, sometimes it does not,

**Screenshot(s)**
[image]

**Actual results**
A gcode file which is 99% correct is generated but fails differently in the end. Sometimes it just moves the print head somewhere and stops, sometimes it moves the head down first and risks crashing into the model. There is no end gcode so the head and bed remains heated.

The screenshot shows an incorrect gcode file and a correct gcode file, both using the same settings and stl. The file sizes are identical but the contents begin to change at the very end of the file (highlighted in the screenshot).

Example ending of a broken gcode file:
[image]

**Expected results**
The gcode should always be generated correctly.

**Project file**
When looking at the slicing result it looks correct, and as mentioned this does not always happen. I include a file where it happened today, but after trying the same project again it correctly created the gcode.
[CCR1_Pontoon reinforcement 2.zip](https://github.com/Ultimaker/Cura/files/4471399/CCR1_Pontoon.reinforcement.2.zip)

Including an example of a broken and working gcode file using the same settings in cura.
[working and broken gcode examples.zip](https://github.com/Ultimaker/Cura/files/4471402/working.and.broken.gcode.examples.zip)

klillas, 2020-04-13:
I can add that I did not used to have this problem. I have been using this cura installation with very similar slicer settings for a long time now. I think it was only about a week ago I started noticing that the prints did not always finish properly.

mahtDFR, 2020-04-14:
Are you soft ejecting the SD card? It's possible it didn't finish writing.
Does the behaviour remain the same if you save a gcode to your computer and then copy it to an SD card afterwards?

klillas, 2020-04-15:
I lack a perfect memory but I try to always use the built in cura eject functionality before physically removing the card from the reader.

I assumed the file is "correctly" written from the point of view of the physical storage because there is ASCII readable contents at the end of the file which differs from the contents of a correct file.

I will play around with different ways of storing the file and removing the SD card to see if I can find a common denominator in the workflow I am using. I will also test with another SD card to see if that somehow is becoming corrupt.

Ghostkeeper, 2020-04-21:
It looks like it continues with a completely different file there halfway through the line. Like the X and Y coordinates are still from one file but then the E coordinate and the X and Y on the next line are from a different file. Since it also has a big jump in the X and Y coordinates on the next line, I mean.

klillas, 2020-04-23:
Could it be that the file was then not completely overwriting the last page of the SSD and it continues using the data from some old GCODE?

Im using the SSD that came with the printer. Only GCODE has ever existed on that card.

If that is the case then I guess the problem was that I did not correctly wait for the SD card to finish before I removed it from the computer, i.e., nothing to do with Cura.

Ghostkeeper, 2020-04-28:
Could be. Once the message pops up in Cura that it saved the file, Cura is done with saving the file. Most operating systems are still writing in the background then though, so you should always eject safely or such issues are going to happen eventually.

closed 2020-04-28 by Ghostkeeper