Ultimaker/Cura #7496: cura setting

abtionlove, 2020-04-14:
The printing setting value changed in Cura 4.2.1 interface will be saved in the place of Cura code. For example, I changed the fill density to 35%, and then where in the Cura code can I see the value I changed? Thank you

rburema, 2020-04-14:
Are you asking "If I change a setting value, where is that reflected in the gcode?" or am I misunderstanding?

abtionlove, 2020-04-15:
> Are you asking "If I change a setting value, where is that reflected in the gcode?" or am I misunderstanding?

No, I'd like to know where to save it to the Cura source code if I change the settings on the interface? And how did it get to the engine? Thank you very much.

mahtDFR, 2020-04-15:
If I understand this correctly you are asking to store a custom default value for infill?
This can be done with a custom profile. Change your infill setting and then choose "Create profile from current settings/overrides..." in the profile drop down in custom mode. If this is not the correct answer, please provide more info by filling out the issue template that you deleted.

Ghostkeeper, 2020-04-21:
If you change the settings in Cura, you will not modify anything in the Cura code, if that's what you're asking.

The changed settings are stored in an Instance Container, which is basically Cura's terminology for "profile". This marks the instance container as "dirty". Upon the next auto-save or upon closing, all dirty instance containers are saved to your configuration folder. The file for the instance container is creating using the [serialize](https://github.com/Ultimaker/Uranium/blob/881a0226c2b9421aae6a0614c181ca8452af141b/UM/Settings/InstanceContainer.py#L441) function.

When starting a slice, all profiles are flattened (so that they are no longer a stack of overrides but just a bunch of settings with their values). They are then put in a message that gets sent to CuraEngine. This all happens in the [StartSliceJob](https://github.com/Ultimaker/Cura/blob/master/plugins/CuraEngineBackend/StartSliceJob.py).

closed 2020-04-21 by Ghostkeeper

abtionlove, 2020-04-22:
> If you change the settings in Cura, you will not modify anything in the Cura code, if that's what you're asking.
>
> The changed settings are stored in an Instance Container, which is basically Cura's [...]

Thank you very much.