Ultimaker/Cura #7499: Minor but annoying UI issues

Sophist-UK, 2020-04-14:
**Application version**
4.5.0

**Platform**
Windows

**Printer**
N/A

**Reproduction steps**
1. Maximise Cura window. Press ESC key. Window is un-maximised - this is unexpected and contrary to normal Windows style.

2. Although there are Maximise/Minimise buttons (top right hand corner of Cura window), and although the system menu has enabled maximise and minimise options when you use the mouse to activate it by clicking on the Cura icon at the top left corner of the Cura window, when you use Alt-Space to activate the system menu the maximise and minimise options are not enabled - which means that those of us who use the keyboard to avoid switching to the mouse and back again cannot minimise or maximise by Alt-Space and then n/m.

**Screenshot(s)**
Can be provided if needed.

**Actual results**
1. Esc de-maximises window.
2. Maximise / Minimise not enabled when system menu is shown using Alt-space.

**Expected results**
1. Esc should not de-maximise window.
2. Maximise / Minimise should be enabled when system menu is shown using Alt-space.

Ghostkeeper, 2020-04-21:
The first issue indeed reproduces. It's weird! I wonder what causes that. In our source code I could find one instance where we're deliberately making the search box unfocus when you press escape, and the account, output selector and print information widgets and all pop-up messages close when you press escape. We're not doing anything with the main window.

The second issue doesn't reproduce for me, but I'm testing this on Ubuntu. Alt+Space opens up a similar menu for me which has maximise and minimise but those work fine for me.

mahtDFR, 2020-04-22:
I could reproduce both of these (Windows 10). Seems a little inconvenient indeed. We discussed in our standup this morning and we will defer this for now, but are open to pull requests to fix it.

Sophist-UK, 2020-04-22:
If I can get Cura to run from source, I will try to fix this.

Misterke, 2021-01-02:
I really hate this behavior.  Wanting to create a new issue for it, I noticed this one, so would love to upvote it!

I so often press ESC as a reflex for de-selecting everything (probably from all other programs using it that way, a.o. Fusion 360) and then Cura again goes to a small window and I have to again maximize it and click on an empty spot on the build-plate.  Yes, it is a small thing, but I'm hitting it so F?!&*! often that I would name it my main annoyance with Cura at this point ...  So, can someone PLEASE have ESC just deselect everything rather than normalizing the window?

64Pacific, 2021-06-02:
With every update I hope that this bug will be fixed - I am always hitting the ESC key to "drop" my Cura tool and this really ruins the user experience.

juandelacruz-calvo, 2021-11-04:
Hiya, I rarely write bugs, but this one is a trap I felt for too many times. If someone can point which line of code does it, I'm happy to write a PR to fix it

juandelacruz-calvo, 2021-11-05:
Thanks @nallath, I'll have a look during the week

Triangulix, 2021-12-05:
I also find the reaction to the escape key annoying. I intuitively press Esc to cancel accidental mouse actions and everytime I end up with the Cura window restored.
I noticed that long time ago but didn't expect it to be a bug. I thought that, like in most CAD software, the developers just believe that UI standards are for noobs.

Combinacijus, 2022-04-02:
Any news on 1. issue?

kosma, 2022-04-10:
I suspect the issue is here:

```python
    @pyqtSlot()
    def exitFullscreen(self):
        self.setVisibility(QQuickWindow.Windowed)
        self._fullscreen = False
```

I think the code should remember the maximized state, be set `QQuickWindow.Maximized` instead of `Windowed` if needed.

goldjunge91, 2022-10-16:
hi this issue get never fixed ?

JosiahBull, 2022-12-29:
This issue is causing me problems on Windows 11!

goldjunge91, 2022-12-29:
I leave from windows to max and no I see it’s a fault from cura team it’s standard MacOS behavior that was somehow shift to windows. But they will never fix the problem if not a customer get trouble 😅

Misterke, 2022-12-29:
Solution is to switch to PrusaSlicer or SuperSlicer ... That fixed it for me ;-)  And once PrusaSlicer has the organic supports in the upcoming 2.6, there's really nothing to miss in Cura anymore ...

goldjunge91, 2022-12-29:
> Solution is to switch to PrusaSlicer or SuperSlicer ... That fixed it for me ;-) And once PrusaSlicer has the organic supports in the upcoming 2.6, there's really nothing to miss in Cura anymore ...

🤣 or in your way but I never get warm with prusa slicer or Slicer 😅but the last week I use bambu studio it’s so much nicer the GUI.
Buuuut with S3D V5 i have paid twice 😅 the fu** hard and deep over years I’m a customer and send them over 10 bugs in v4 and no I have to pay again but still better as anything on the market and they got me to pay twice full 😅🤣

NikAWing, 2023-06-10:
3+ years ... let's hope for a fix in v5.4 ... :)

very OT here: (though 5.3.1 has other weird bugs that annoy you when you try to rotate objects or select surface which you want to be bottom)

Ron-ski, 2023-11-26:
This bug still exists in 5.5 and its so annoying, I hit esc expecting it to deselect, but nope the window changes size!!!! Very annoying as its a reflex action o hit escape. This is on Windows 10 Pro.

onolox, 2024-01-11:
It's a real shame to Cura devs that a simple bug like this is not corrected. 
Is doing anniversary.

Sophist-UK, 2024-01-19:
I took a look at the community submitted PR #14356 and I am **not** surprised that it wasn't merged as it has far too many changes that are unnecessary to the fix and which will likely have unintended consequences (like removing whole menu items) rather than just stopping the Esc key from doing something.

I have resubmited a new 1-line change PR #18081.

PR #18081 (2024-01-24): Stop Esc key from exiting full screen mode
# Description

Fixes #7499

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Printer definition file(s)
- [ ] Translations

# How Has This Been Tested?

It hasn't been tested - because I was previously unable to create a run-from-source environment.

# Checklist:

- [x] My code follows the style guidelines of this project as described in [UltiMaker Meta](https://github.com/Ultimaker/Meta) and [Cura QML best practices](https://github.com/Ultimaker/Cura/wik [... cut]
closed 2024-01-24 by nallath