Ultimaker/Cura #7500: Extrusion Instead of Retraction in Filament Change

suryanisaac, 2020-04-14:
**Application version**
Cura 4.5
**Platform**
MacOS 10.15.4

**Printer**
Ender 3

**Reproduction steps**
1. Extensions on Top Menu
2. Post Processing
3. Modify GCODE
4. + -Filament Change
5. Set retraction to 50mm

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
50mm of filament is extruded instead of retracted, causing stepper slips and ruining the print

**Expected results**
50mm of filament should be retracted

**Project file**

**Log file**

**Additional information**
The problem was resolved when -50 was entered in the retraction field. I think the person who coded this didn't have enough coffee and forgot to added the '-' before the extrusion amount :)

closed 2020-04-14 by suryanisaac

Ghostkeeper, 2020-04-21:
This has been changed back and forth a number of times:
* https://github.com/Ultimaker/Cura/commit/c7b146c9fc3bc287b456e9eea5b17aad4b689086
* https://github.com/Ultimaker/Cura/commit/296b74b03244ee4ca42c934c8feb43c6c69bb76c
* https://github.com/Ultimaker/Cura/commit/5592d4fa602835536ea7b57778c7fd744294a182

We kept changing it since the Ultimaker printers don't support this g-code so we can't test either way. However when looking at the history of that minus sign there, I must conclude that different firmware interprets this differently.

Looking at documentation, the [RepRap wiki](https://reprap.org/wiki/G-code#M600:_Filament_change_pause) sort of words it ambiguously. The [Marlin Documentation](https://marlinfw.org/docs/gcode/M600.html) clearly says it should be negative, however the [actual Marlin code](https://github.com/MarlinFirmware/Marlin/blob/5cf6c44f47737bdf93df6ae7c835fe2d54d0675d/Marlin/Marlin_main.cpp#L10993) wraps this value in `ABS()` and so whether it's negative or not shouldn't matter to Marlin.

suryanisaac, 2020-04-22:
> c7b146c <https://github.com/Ultimaker/Cura/commit/c7b146c9fc3bc287b456e9eea5b17aad4b689086>
> 296b74b <https://github.com/Ultimaker/Cura/commit/296b74b03244ee4ca42c934c8feb43c6c69bb76c>
> 5592d4f <https://github.com/Ultimaker/Cura/commit/5592d4fa602835536ea7b57778c7fd744294a182>

Yeah, different firmwares might interpret it differently. My Ender 3 runs the Creality modified firmware (might be a derivative of marlin).  I’ll burn a boot loader and try it with marlin soon.