Ultimaker/Cura #7501: GCODE editing box not easily accessible

suryanisaac, 2020-04-14:
**Application version**
Cura 4.5

**Platform**
MacOS 10.15.4

**Printer**
Ender 3

**Reproduction steps**
1. Settings on top bar
2. Printer
3. Manage Printers
4. Machine Settings

**Screenshot(s)**
[image]

**Actual results**
The GCODE editor on the bottom doesn't render properly

**Expected results**
The GCODE editor should be taller

**Project file**

**Log file**

**Additional information**

nallath, 2020-04-15:
Could you try again with the 4.6 beta? We've made some changes there (for instance; it should be possible to make that window bigger by dragging).

If that doesn't work, we might just be in time to fix it for the 4.6 release!

suryanisaac, 2020-04-15:
Yup, it's the correct size now, but I still can't resize it by dragging. You guys might want to look into that.

keegreil, 2020-04-20:
I thought I had this issue to yesterday, could not see the G-code boxes to edit, and could not resize the window.  Thought it was related to my screen resolution (4k monitor) and Windows 10's "Scale and Layout" display settings.  At 150% the boxes were not visible at all.  At 100% they looked fine.  Today I went to take screenshots for this report, and now the boxes are resizable, so it works at either 100% or 150%.  Cura 4.5 on Windows 10 64-bit. Don't know if that helps you at all.  
[image]

Ghostkeeper, 2020-04-22:
> Yup, it's the correct size now, but I still can't resize it by dragging. You guys might want to look into that.

We've seen this before. It was also reported here: https://github.com/Ultimaker/Cura/issues/7300 Apparently a MacOS-specific issue because on Windows and Linux people are able to resize it.

closed 2020-07-06 by no-response[bot]