Ultimaker/Cura #7503: [Dual Extrusion] Shell on E1 and support on E2 work, but not the other way round

ScruffR, 2020-04-14:
**Application version**
4.6.0 beta

**Platform**
MS Suface Pro 2017 (Intel Graphics)
Windows 10 Pro 64bit (Build 19555.1001)

**Printer**
Creality CR-10 (dual extrusion mod with E3D Chimera)
**Printer**

**Reproduction steps**
1. select object and assign to extruder 1
2. set support extruder to extruder 2
3. slice and preview - works as expected
4. reload the project but ...
5. select object and assign to E2
6. set support to E1
7. slice and preview - the object is now considered solid

**Screenshot(s)**
preview after 3.
[image]
preview after 7. 
[image]

**Actual results**
Where there previously was support (zig-zag) directly touching the build plate there is a solid bottom layer and infill (gyroid) instead of white support

**Expected results**
It should be the same as the first screenshot (after step 3.) but with inverted colours (yellow shell, white zig-zag support)

**Project file**
[StarWithEyes.zip](https://github.com/Ultimaker/Cura/files/4476898/StarWithEyes.zip)

**Log file**
n/a

**Additional information**
n/a

Ellecross, 2020-04-14:
Can you check in extruder 2 that "Remove all holes" is disabled?

ScruffR, 2020-04-15:
Duh' 🤦‍♂️ 
Thanks, that was the issue 😳 

I forgot to check for that. 
Some settings impact the whole project and others only pertain to the selected extruder - I never know which and why.
Would be nice if the gerneral once could get some kind of marker - or even get their own tab when working on a multi-extruder printer.

closed 2020-04-15 by ScruffR