Ultimaker/Cura #7504: Print brim from inside outwards

Sophist-UK, 2020-04-15:
**Is your feature request related to a problem? Please describe.**
At present the brim is printed from outside to inwards, and because first layer is squashed, sometimes the width of the line is wider than it should be and this causes overlaps or (if the printer has some laxness) pushes the head slightly sideways. Because the brim is printed inwards, these accumulate and eventually impact the first layer of the object itself.

[image]

Since the brim is sacrificial, if we printed the brim from inside outwards, these accumulated errors would not impact the object itself.

Note: These overlaps are actually better avoided by e.g. reducing the material flow for brims slightly (I used 95%) as they weaken the brim adhesion.

**Describe the solution you'd like**
Print brim from inside outwards rather than outside inwards.

**Describe alternatives you've considered**
Adjust material flow for brim to avoid overlaps.

**Affected users and/or printers**
Everyone

Ghostkeeper, 2020-04-22:
You can use a skirt with 0 distance and multiple lines?

At this point, brim and skirt are pretty much equal aside from a few details and defaults.

But really, I think the real solution is to fix the flow rate. Maybe it's printing too hot.

Sophist-UK, 2020-04-22:
I have achieved perfect brims since then - maybe I had the flow rate a little high and it was overlapping.

We can close this if you want.

Sophist-UK, 2020-08-31:
On reflection, brims should be printed outside in as part of their role (in addition to helping prevent the part breaking away) is to allow extrusion flow to settle (like a skirt) - and for this it needs to be outside inwards.

closed 2020-08-31 by Sophist-UK