Ultimaker/Cura #7505: Build Plate temperature initial layer

giantshell, 2020-04-15:
**Application version**
cura 4.5

**Platform**
OSX 10.15.4

**Printer**
FLSUN QQS

**Reproduction steps**
1. define settings in cura
2. slice the settings in cura
3. Save on HD
4. Open created file on HD and check instruction code M190 and M140

**Screenshot(s)**
(Image showing the
[image]
 problem, perhaps before/after images.) 
[image]

**Actual results**
The M190 instruction at the biginning of Gcode does not refer to "Build Plate temperature initial layer" parameter but to "build plate temperature" parameter

**Expected results**
The M190 instruction at the biginning of Gcode should refer to "Build Plate temperature initial layer" parameter and M140 should refer to "build plate temperature" parameter

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-04-22:
Yeah, this was a mistake in the profile for FLSUN QQ-S that we merged in a year ago. Thanks for spotting it! I can't test this profile, but it's an obvious mistake and an easy fix so I put it in the next release myself anyway.

closed 2020-04-22 by Ghostkeeper

giantshell, 2020-04-22:
Thanks, Good news ;-)

giantshell, 2020-04-29:
Hi,

I’ve checked in V4.6 of cura and problem remains.

Any information ?

Thanks.

Ben

Ghostkeeper, 2020-04-30:
Yeah the change was made to master, since 4.6 was released on the day before.

commit 888937d0 (2020-04-22): Use initial layer bed/print temperatures

These are more likely what the user wants the printer to start up to.

Fixes #7505.