Ultimaker/Cura #7506: Slicer is filling in holes

men8ifr, 2020-04-15:
[CCR10_half monitor mount.zip](https://github.com/Ultimaker/Cura/files/4480352/CCR10_half.monitor.mount.zip)

**Application version**
4.5

**Platform**
Win 10 Core I7 Geoforce GTX

**Printer**
CR10

**Reproduction steps**
1. Made a simple model in Fusion
2. Slice it 

**Screenshot(s)**
[image]

[image]

**Actual results**
Cura fills the holes in - it should not

**Expected results**
The sliced model should have a hole through it - as per the STL

**Project file**
Zip file Should be uploaded 

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues
[cura.log](https://github.com/Ultimaker/Cura/files/4480409/cura.log)
 to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
There seem to be quite a few similar issues with no clear resolution I can see, if it's related to my settings it doesn't seem obvious

men8ifr, 2020-04-15:
[cura.log](https://github.com/Ultimaker/Cura/files/4480469/cura.log)

nallath, 2020-04-15:
I'm still investigating a bit more, but your model doesn't seem to have any issues with it (my first guess was that it was non-manifold). But when resetting all the settings to default, it does print the holes correctly.

nallath, 2020-04-15:
Ah! Found it! You had "Make Overhang Printable" enabled. So it's to be expected that this happens.

closed 2020-04-15 by nallath

men8ifr, 2020-04-15:
Many thanks. How did you compare my profile to the standard? Maybe that is a good way to debug other issues in the future. It is easy to get lost in the many options available under all the settings.

Ghostkeeper, 2020-04-22:
It lists the changes in a profile when you open the profile manager (the last page of the Preferences screen) and select the profile you want to view.