Ultimaker/Cura #7508: Speed sensitive flow compensation

men8ifr, 2020-04-15:
**Is your feature request related to a problem? Please describe.**
When I print faster,  I get under-extrusion. If I Increase the flow multiplier I get over-extrusion at corners where the print head is moving slower. 

**Describe the solution you'd like**
A factor that increases the flow % with increasing speed so that e.g at the beginning of a line where printhead speed is low the flow = 100%, in the middle of the line when the printhead is at maximum speed e.g. 50mm/s the flow is set to something the user defines - e.g 110%

**Describe alternatives you've considered**
This is basically the same as linear advance expect including it in the slicer. Many users won't set-up and configure linear advance due the difficulty implementing it, even though it is well recognised to improve print quality. Further many new printers with silent steppers cannot implement linear advance as it is not compatible with the silent driver boards.  

**Affected users and/or printers**
Every-one but bowden based printers suffer worse. 
[image]
Attached photo is only at max 50mm/s

**Additional context**
See https://marlinfw.org/docs/features/lin_advance.html

Ghostkeeper, 2020-04-22:
You can configure Linear Advance from Cura as well by putting it in your start g-code, or by installing the Linear Advance plug-in from the Marketplace which adds a setting for it.

Cura also has an implementation similar to this called [Flow Rate Compensation Factor](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/flow_rate_extrusion_offset_factor.md) in Experimental. It never really worked well for us at Ultimaker, since we have printers with Bowden tubes. We essentially gave up the experiment and don't really plan on continuing that pathway any more. If someone would like to continue that implementation though they'd be welcome to give it a try.

64Florian, 2020-05-07:
>  Attached photo is only at max 50mm/s

Have you try to "overheat" a bit your head / filament? 
In my point of view, if you head  is not hot enough, you will increase the force in the tubes and when you slow down, you will reduce the pressure but have an overload of materials
What kind of extruder have you on your printer ? (I use a Titan extruder on a Tevo Tarantula and I work at 210 / 215 °c for a 80mm/s - my PLA is a 190-220°c)

Where could I have a sample of you design to see if I have the same issue with my printer ?

Ghostkeeper, 2022-07-18:
Something like this has been implemented in Cura 5.0. But it only applies to variable line width, because the flow differences in the different structures (e.g. infill vs. outer wall) need to adhere to the speeds set for these structures.

ansonl, 2023-07-29:
@Ghostkeeper Is [Flow Rate Compensation Factor](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/flow_rate_extrusion_offset_factor.md) still not suitable for bowden/Ultimaker setups? 
The setting is still set to 100% by default in the Ultimaker configurations so should it actually be set to 0% if we use another type of linear advance method to avoid conflicts? Thanks.