Ultimaker/Cura #7509: Outer before inner walls

Zdendo463, 2020-04-15:
Can you add function "Outer before inner walls" only for the first layer? It is much better, if the model have some small holes. If it print this small objects, they are unsticking from the bed. If I print "outer before inner walls" - It will printing first long outer paths, and then it printing small "holes" connected to outer wall, and it is sticked properly onto bed. The problem is, if I need print overhangs. So - great solution is - "Outer before inner walls only for the first layer" function. Or something like this. Thank you.

Ghostkeeper, 2020-04-22:
Wouldn't the outer walls be *smaller* than the inner walls for those small holes?

We already have special rules for the outer vs. inner walls on the initial layer though. This is to prevent the nozzle skimming over previously printed parts. Essentially when you're printing with a brim it'll do the outer wall first. When you're printing with a skirt it'll do the inner walls first.

64Florian, 2020-05-24:
Hello,
For me I see this usage:
If you want to print a grid 45° angle, it will be better to start on nice and continue print as base.

[image]

 And for other layers you need to have the inner wall to "stick" the outer wall (prevent it to fall)
[image]
[sampleOuterBeforInner.3mf.zip](https://github.com/Ultimaker/Cura/files/4673522/sampleOuterBeforInner.3mf.zip)