Ultimaker/Cura #7510: Z height correction

raoul596, 2020-04-15:
**Is your feature request related to a problem? Please describe.**
If I print a part with a height of 60mm using 0.25mm layer height and an initial layer height of 0.1mm the part will only be printed 59.85mm high(Cura shows 240 layers so 239layers*0.25mm+0.1mm = 59.85mm)

**Describe the solution you'd like**
Is there a possibility for a new feature like z height correction, so it will print a final layer on top of 0.15mm, so the part will be 60mm high.

**Describe alternatives you've considered**
A nice alternative is using the setting: 'adaptive layers' which will compensate the last layers of the print so that it will finish on exactly 60mm in my case.

**Affected users and/or printers**
Everyone would benefit from this feature since all prints will have the right Z height.

raoul596, 2020-04-16:
The same would apply to the setting 'support Z distance'. It would be really nice if you set support Z distance to, for example, 0.2mm and the gap will always be 0.2mm, even when printing at layer heights of 0.15mm or 0.25mm

Ghostkeeper, 2020-04-22:
There are a couple of problems with this feature. I'll try my best to explain. I'm going to use smaller numbers here and assume the initial layer height is the same as the normal layer height for the simplicity of my argument. Let's say you have a layer height and initial layer height of of 0.25mm.

Let's say that you want to print a cube of 1.1mm. Currently this would produce 4 layers of 0.25mm, because the 5th slice would be at a height of 1.125mm which doesn't intersect with the cube. But we could make another slice at 1.05mm and print that at a lower rate (barring the third argument below). But now the user places a 2.0mm cube next to it. You'd still want to print the first cube in the same way, but now you have a layer of 0.1mm in between so you're going to have a band of 0.1mm running through the middle of your second cube. And also you'd again run into the problem that your second cube then prints 8 layers for a total of 1.85mm height and then needs to print a 9th layer.

Secondly, extending the first argument, what would happen if the 2 cubes are merged together as one mesh forming a small stair case. You still want the smaller cube to be accurate, but you'd also want the taller cube to be accurate. If you make such a stair case with many steps you'd have many bands of smaller layer heights running through it. Worse, if you're printing something with a slope you'd have an infinitesimal layer height.

And thirdly, but separately, you'd have a problem with flow rate. Adaptive Layers has a routine that limits the maximum change in flow that you can have between two layers, because if you switch from a 0.25mm layer to a 0.1mm layer then you're suddenly printing at 40% flow. The current pressure in the nozzle chamber is too high for that so you'll be overextruding for a while. And if there's another object next to it, you'll be underextruding for a while on that object, creating thin horizontal gap in its surface. You'll also run into surface tension problems if you have a layer of 0.02mm thick that needs to be extruded with a 0.8mm nozzle. This problem can be solved with a minimum layer height, but then your height will be inaccurate again.

Closely related, all of the other settings will be tuned to a higher layer height. For instance, typically with a larger layer height you'll also need a higher temperature to get the material molten in time due to the higher flow rate. This could cause the material to degrade and the nozzle to clog if your highest layer is large.

So I'd like to open this for discussion.

As a workaround, you can adjust the "Slicing Tolerance" setting to get to a rounding that fits your purpose better.

64Florian, 2020-05-24:
Hello everyone,
As "quick fix" it could be interesting to rise a warning and ask the user "Your product height doesn't fit to your model" 
with 3 answers
"Keep my settings" - "Adapt for thicker layer (same print time)" - "Adapt for Thinner layer (add one layer / slower print) "
This feature could be enable or disable as "Auto preserve product height"

Of course, few point raised by Ghostkeeper stay opened... but for "basics prints" it could be useful...
My point of view is "If the user can have in mind, the layer height can change the final product height, the use will be able to take in account this and prefer print always at 0.2mm than 0.25 if an 0.1 mm is it's max tolerance " 

hem... 
In fact, the "Z product tolerance"  idea could be nice to have too...
-> the user set this value and Cura rise a warning if something is "bigger" to this "tolerance" due to user settings (**Mandatory for "professional feature / production" in my point of view**)

GregValiant, 2026-09-10:
I'm going to go ahead and close this.

closed 2026-09-10 by GregValiant