Ultimaker/Cura #7512: Infill minimum gap

Sophist-UK, 2020-04-15:
**Is your feature request related to a problem? Please describe.**
When making an engineering part there is a compromise to be made between strength and weight and different areas require different infill.

In particular, narrow areas may require 100% infill (with skin?) in order to give it compression strength e.g. to resist compression from a bolt/nut, whilst the main body could e.g. be 30% Cubic Subdivision to give it overall strength but lightness.

At present, the only control in this area is Minimum Infill Area, which is good as far as it goes, but doesn't distinguish between small self-contained areas (like legs, the example in the settings guide) and small but connected areas which still need 100% infill but because they are connected to other bigger infill areas do not trigger the Minimum Infill Area fix.

**Describe the solution you'd like**
I would like a Minimum Infill Distance setting whereby any infill point where a circle of this diameter intersects walls is filled 100% (with skin?) rather than having normal infill.

**Describe alternatives you've considered**
None.

**Affected users and/or printers**
Users printing parts for machinery where strength of narrow areas of the object is important.

Sophist-UK, 2020-04-15:
P.S. I am not sure what the computation cost is for computing whether a circle intersects with walls, but it should be possible to produce some sort of optimisation e.g.

1. Having determined that a point is or isn't intersected, a test is only done when the head has moved a % of the Minimum Infill Distance setting. 

2. Further optimising the above, whereby if this circle is not intersected, then a similar check for a circle double the diameter is made so that if that is negative then a larger distance can be skipped before doing another check for the larger radius.

P.S. As an alternative, if Cura identified potential areas, then you could have a tool for the user to click and select such areas for 100% infill.,

smartavionics, 2020-04-15:
Sounds a bit like the gradient infill feature that has recently been all the rage (see https://github.com/smartavionics/Cura/issues/38).

Sophist-UK, 2020-04-15:
I have lost the reference to it for the moment, but I was chatting with someone who has a fork of Cura that includes Gradient Infill functionality that he added and which was apparently rejected as a PR by Unimaker. :-(

But no - this is NOT Gradient Infill - this is to make some small areas which need to be solid for engineering reasons.

smartavionics, 2020-04-15:
Actually, the gradient infill feature has not been put into a PR, it's just in my Cura builds.

Anyway, the point I think I was making was simply that, like gradient infill, your suggestion concerns the modification of the infill due to the proximity of the outside world.

Sophist-UK, 2020-04-15:
Yes - it has that in common.

Ghostkeeper, 2020-04-22:
> I would like a Minimum Infill Distance setting whereby any infill point where a circle of this diameter intersects walls is filled 100% (with skin?) rather than having normal infill.

This is actually not very difficult to do. What you said is just an inset of the infill area, but what you meant is called an [opening operation](https://en.wikipedia.org/wiki/Opening_(morphology)) in morphology. The explanation on that Wikipedia page is focused on image processing rather than geometry but the picture makes it clear. It's done by making an inset and then an offset, both of which we have methods for from Clipper. It's not the most efficient operation, but we generally do dozens of them per layer right now. We currently perform an opening operation on the skin areas to remove thin parts of skin (as per the Skin Removal Width setting).

Unintended side-effect is though that you'd also get skin in all of your outside corners, or at least the sharpest corners (dull corners being excluded due to being too small to print with the skin line width). These are typically not the place where you want to have skin since by nature of their geometry the outside corners are already the strongest points of the print.

There is also the argument to make that it may not be common for prints to want specifically their thinner parts to be stronger. Why would thinner parts need to be stronger for a generic print? We already provide an option for *any generic part* of your model to be printed with higher infill rate. But it's just harder to use.

Sophist-UK, 2020-04-22:
I got round this by increasing the skin thickness substantially to give both strength in small areas like clamps but also strength to large bodies (think Brunnel's box girders).