Ultimaker/Cura #7513: Print to USB/Network - Display layer and M117 LCD messages

Sophist-UK, 2020-04-15:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
In addition to the Job Name, Printing time and Estimated time left, also display:

a. The current Layer Number / Total Layers
b. Any M117 LCD Messages sent to the printer

**Describe alternatives you've considered**
None.

**Affected users and/or printers**
USB / Network printers.

Ghostkeeper, 2020-04-22:
We have a post-processing script that does this, actually! It's called "Display Filename and Layer on LCD". It shows the job name, the current layer and total layers. It can optionally also display a customised message with it.

closed 2020-04-22 by Ghostkeeper

Sophist-UK, 2020-04-22:
I apologise if I did not explain this clearly. The post-processing script(s) that you refer to show the information on the printers screen - this Enhancement request suggests that it is also shown on the Cura tab for the USB / Network print.