Ultimaker/Cura #7517: [4.5.0] Pause at height, redo layers inconsistent

azhural, 2020-04-16:
**Application version**
4.5.0

**Platform**
Ubuntu 18.04

**Printer**
Creality Ender-3

**Reproduction steps**
1. Use Pause at height
2. Set to layers, choose any layer
3. Set redo layers to 10, 1 or 0.
4. Slice

**Actual results**
1. Redo layers: 10
At layer 50 it moves to Z10.2, does the pause, then prints layer 40 at 10.2. Then moves to 8.4 to do layer 41.
```
;LAYER:49
;TYPE:WALL-INNER
;MESH:xyzCalibration_cube.stl
G1 F1500 X108.5 Y126.5 E750.57301
G1 X108.5 Y108.5 E751.17169
G1 X114.995 Y108.5 E751.38772
G1 X115.001 Y110.397 E751.45081
G1 X118.273 Y110.397 E751.55964
G1 X118.882 Y109.767 E751.58878
G1 X118.884 Y108.5 E751.63092
G1 X126.5 Y108.5 E751.88423
G1 X126.5 Y114.858 E752.0957
G1 X124.564 Y114.859 E752.16009
G1 X124.563 Y119.326 E752.30867
G1 X126.5 Y119.33 E752.37309
G1 X126.5 Y126.5 E752.61157
G0 F9000 X126.9 Y126.9
G1 F1500 X108.1 Y126.9 E753.23686
G1 X108.1 Y108.1 E753.86215
G1 X115.394 Y108.1 E754.10475
G1 X115.4 Y109.997 E754.16784
G1 X118.103 Y109.997 E754.25774
G1 X118.482 Y109.605 E754.27588
G1 X118.484 Y108.1 E754.32593
G1 X126.9 Y108.1 E754.60585
G1 X126.9 Y115.258 E754.84393
G1 X124.964 Y115.259 E754.90832
G1 X124.963 Y118.927 E755.03032
G1 X126.9 Y118.931 E755.09474
G1 X126.9 Y126.9 E755.35979
G0 F9000 X127.3 Y127.3
;TYPE:WALL-OUTER
G1 F1500 X107.7 Y127.3 E756.01169
G1 X107.7 Y107.7 E756.66359
G1 X115.793 Y107.7 E756.93276
G1 X115.799 Y109.597 E756.99586
G1 X117.933 Y109.597 E757.06684
G1 X118.082 Y109.443 E757.07396
G1 X118.084 Y107.7 E757.13194
G1 X127.3 Y107.7 E757.43846
G1 X127.3 Y115.658 E757.70315
G1 X125.364 Y115.659 E757.76754
G1 X125.363 Y118.528 E757.86296
G1 X127.3 Y118.532 E757.92739
G1 X127.3 Y127.3 E758.21901
G0 F9000 X127.03 Y127.03
G0 X126.419 Y124.904
;TYPE:FILL
G1 F3000 X110.094 Y108.579 E758.98689
G0 F9000 X112.569 Y108.578
G1 F3000 X108.579 Y123.467 E759.49957
[... 531 lines cut ...]
G0 F9000 X108.579 Y123.836
G1 F3000 X111.162 Y126.419 E772.90576
G0 F9000 X112.074 Y126.419
G1 F3000 X126.418 Y122.576 E773.39967
G0 F9000 X124.483 Y116.883
G1 F3000 X108.58 Y121.144 E773.94726
G0 F9000 X108.579 Y115.351
G1 F3000 X119.648 Y126.419 E774.46789
G0 F9000 X120.066 Y126.42
G1 F3000 X124.846 Y108.579 E775.08221
G0 F9000 X126.419 Y110.152
G1 F3000 X108.579 Y114.932 E775.6965
G0 F9000 X118.126 Y110.475
G1 F3000 X113.854 Y126.419 E776.24551
G0 F9000 X124.858 Y114.659
G1 F3000 X118.921 Y108.721 E776.52479
;MESH:NONMESH
G0 F300 X118.921 Y108.721 Z10.4
G0 F9000 X126.5 Y126.5
;TIME_ELAPSED:1033.987623
;LAYER:51
```

**Expected results**
I interpreted "redo" as "print the exact same layer again", which seems to be true from the second redone layer onwards (see redo 10). The first layer redone is at the wrong height imo.

Ghostkeeper, 2020-04-22:
The intent is that it redoes the last layer before the pause because the act of pausing might bring about some ooze, so you'd get some underextrusion just after restarting the print. You would "prime" by printing one extra layer. This is necessary because other methods of priming are not feasible (priming on the build plate can't be done since the build plate might not be accessible any more, priming on the prime tower might not be done because no prime tower is printed up to the current height, and priming in the infill can't be done because at this stage of processing it's unknown what area is filled with infill).

Why you would ever want to redo more than 1 layer is unknown to me. It seems downright dangerous because the nozzle could clash into the previously printed parts. Should it maybe not make any Z moves at all in the redone part?

azhural, 2020-04-22:
I was not planning on redoing 10 layers, I was trying to figure out the intended behavior.
The way you described it, a toggle would convey that intention better than a freely selectable number.
Also it's not really an _extra_ layer, as the redone layer and the actual next layer are at the same height.

> Redo a number of previous layers after a pause to increase adhesion.

The tooltip suggests redoing multiple layers might be a usecase (although I can see your point that it probably is not), so I believe my initial point still stands: The actual behavior is all over the place. The first redone layer is added as an extra layer with all the following layers way below inside the already printed part. The part after the redo then starts at the same height as the extra layer.

So if the issue is not "inconsistent behavior" then it should be "UI does not convey intention".

Sophist-UK, 2020-04-24:
I would propose the following changes:

1. UI should have a checkbox for Redo current layer rather than a numeric box.

2. It seems to me that the redo of the current layer should be more like an iron (using the same settings i.e. slower flow, avoid outer wall. The current layer has already been extruded, so you don't want to try to extrude a full layer of filament, but instead to extrude a much smaller amount that will be pressed firmly into the current layer.

Ghostkeeper, 2020-05-22:
We've implemented point no. 1 for the next release. With that, the issue here is hidden and no longer in scope.

We did not implement Sophist-UK's second suggestion though. The purpose is not to flatten the surface of the previous layer, but essentially to:

1. Prime the nozzle after being paused and hot and oozing for a while.
2. Ensure sufficient connection after the pause. Pausing often causes a fault line, due to the material cooling down too much and getting time to warp. For some materials it's better then to overextrude a bit to serve as a glue layer.

closed 2020-05-22 by Ghostkeeper