Ultimaker/Cura #7518: "Oops" - minor change in flight at layer change

Sophist-UK, 2020-04-16:
**Is your feature request related to a problem? Please describe.**
I don't know about you but quite often I start a print and realise 30s into a USB print that I should have tweaked a setting or perhaps 5m into the print I realise from the quality so far that I need to tweak a setting and would be happy for the tweak to happen from there on in rather than clean the bed and start again from the beginning.

So wouldn't it be nice if...

We could reslice and then tell the Monitor tab to switch to the current slice at the end of the current layer.

**Describe the solution you'd like**
A button on the Monitor tab that says Replace Slice at end of Layer. When you click the button Cura prints to the end of the current layer and then switches to the current version of the slice, reading through it until it gets to the comment that indicates the start of the next layer.

It would probably need to capture Gcodes that indicate a permanent change like hot-end temperature, fan speed etc. and replay the last of each prior to starting the next layer just in case the tweak was e.g. changing the temperature.

**Describe alternatives you've considered**
None.

**Affected users and/or printers**
All.

Ghostkeeper, 2020-04-22:
I'm going to have to defer this feature request. We don't intend to progress the development of USB cable printing further. The reason for that is because Ultimaker only supports USB cable printing with the Ultimaker Original which is no longer being produced. We can't justify spending time on a feature such as this. If someone else takes the effort to make this sort of thing happen we'd be happy to give it a review.

I think we'll also need to be careful to not completely ruin a print after making the swap. Generally Cura should prevent damage to the printer (best effort), and crashing into previously printed parts can result in damage. Maybe it'd be best to only allow changes such as the ones allowed on the printer's display, like speed, temperature, fan speed.

Sophist-UK, 2020-04-22:
Yes - of course the user would need to be careful only to do this when the slices remain compatible.

So changing the objects being printed, moving them on the bed, changing the scale, changing the layer size are all things that would cause issues. Ideally Cura would disable the Oops button if the user did any of these things.

I have not used Network printing, but if that uses the same functionality as USB printing then this might still be worthwhile.

P.S. I appreciate that you are primarily interested in functionality for Ultimaker printers, but it would be very disappointing if you ignore the vast majority of Cura users who do not have Ultimaker printers. That said, I think the community should be the developers of enhancements that don't benefit Ultimaker printer owners, just so long as Ultimaker is positive about accepting and including such PRs.

Ghostkeeper, 2020-04-28:
We are accepting of such PRs as long as it doesn't make the application too complex so that it's unmaintainable.

The network printing that Ultimaker printers implement sends a print to the printer and then Cura is done. Cura could be closed down as it has nothing to do with that print any more. There is a monitor page that allows tracking the current state of the printers you're connected to, even if someone else sent the prints or your prints are still waiting in a queue. There is no way to "download" the current g-code from the printer and then change stuff in it, so this feature doesn't really apply to those printers.

Sophist-UK, 2020-04-28:
It won't be too complex. The approach I currently imagine is as follows:

1. When you start a USB print enable the "Replace Slice" button.

2. When you do something that would make a new slice incompatible - disable the "Replace Slice" button. This is the complex bit - working out what actions might make the slice incompatible. Examples are: adding or deleting an object, moving or scaling the object, changing the layer height, changing the infill pattern or % (if you have printed past the bottom thickness, changing the support settings - but there are likely to be others.

3. When you replace the slice, then you find the beginning of the next layer in the current slice. Then you read the new slice until you get to the beginning of the same layer (which will be in the same position due to the above constraints) - and as you read the file you look for gcode which sets the machines state and apply onwards into the print - temperatures, speeds - there will be a defined list, including some which do the opposite of others and you need to save the last one. You save the most recent of each of these and replay them before switching. You might also have to do something special about absolute extrusion measurements or relative position measurements.

4. Then you switch. slices.

Ghostkeeper, 2020-04-29:
That seems like a reasonable approach. Hopefully it could be contained within the USB printing plug-in (aside from maybe needing a generalisable extra signal or something; that would be acceptable).

closed 2023-07-13 by github-actions[bot]