Ultimaker/Cura #7520: Initial Layer Print Speed ignored

iammenotyou, 2020-04-16:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-04-22:
I'm unable to reproduce this issue. Here is a screenshot where I loaded a cube and set the print speed to 70mm/s and initial layer speed to 20mm/s:
[image]

As you can see, the first two layers have a darker blue colour than the rest, indicating a lower speed. It's two layers because the Ender 3 by default sets "Number of Slower Layers" to 2.

Maybe you have a project file that reproduces this problem?

64Florian, 2020-04-27:
Hello,
@iammenotyou : Could you made a Gcode as sample please ? 
accoring your writings this issue is seen on your printer not in the simulator.
Could you check if in the simulator (as done by @Ghostkeeper) you have 2 kind of colors in you side too  ? 

I noticed something when I try with my Mac version : 
If the skirt/brim speed is manually set it can "appear" for the user this kind of effect (but for my test the speed for the 1st layer is the "good" one)
[image]

I hope it can help

Ghostkeeper, 2020-04-28:
@64Florian That's correct. By default the value for skirt/brim speed is set to be equal to Initial Layer Speed, but you can override it to something else if you like.

iammenotyou, 2020-04-29:
@64Florian  [https://drive.google.com/file/d/1QVgU27HsmiGNAyLjjui4Q2fj95rC-n45/view?usp=sharing](url) I've attached a sample gcode file.   Its sliced for an ender 3 pro at 20 mm/s first layer and 70mm/s 2nd layer.
 
On the weekend I'm going to try to generate a gcode with another slicer just to confirm it is the slicer, because I've found the Marlin 2.0 has issues with ABL (does the motions just doesn't adjust the height during printing) maybe it has ignores speed, maybe it also ignores print speed.   Will update on the weekend.

64Florian, 2020-04-29:
Thank you @iammenotyou, 
According your file for layer 0 you have (I have added explanation) : 

----------------------------------------
;LAYER:0
M107 -> Fan Off 
G1 F300 Z0.5 -> go to 0.5 in Z axes @ 300mm/min (5mm/s) (Linear Move)
G0 F6000 X54.148 Y111.073 Z0.5 -> go to X54.148 Y111.073 Z0.5 in Linear Move @ 6000mm/min (10mm/s) = Move to the start position
;TYPE:SKIRT -> description of what will be done
G1 F300 Z0.2 ->  go to 0.2 in Z axes @ 300mm/min (5mm/s) (Linear Move) = move down from 0.5 to 0.2
G1 F1800 E0 ->  go to 0 for extruder @ 1800mm/min (30mm/s) (Linear Move)
G1 F1200 X54.468 Y110.946 E0.00916 ->  go to X54.468 Y110.946 E0.00916 @ 1200mm/min (20mm/s) (Linear Move)
G1 X54.819 Y110.843 E0.01889 -> go to X54.468 Y110.946 E0.00916  (Linear Move) => Work, (and keep the same speed )
------------------------------------------------

According those information the Gcode is "good" for me ... 
But your printer could not manage (may be ?) speed and X+Y+E (but strange...)

If you want to try, I have change your file and keep only SKIRT and 1st layer ... 
I have modify the instruction to try to change the "fast" motion (travels in fact) to 10 time slower (from 6000 to 600 mm/min 100mm/s to 10mm/s)
[salt_stand_64Florian.gcode.zip](https://github.com/Ultimaker/Cura/files/4555188/salt_stand_64Florian.gcode.zip)
You should "ear" the difference instantly ! (and see: realy slow motion)

If it's works as expected, it show: 
 - Your printer was not able to handle  speed+X+Y+E 
If nothing change (as your mention) it means it's something else must be tuned in the file (I have an Idea with the "G1 Z2.0 F3000 ;Move Z Axis up" but I don't want to mix-up 2 tests at the same time)  

I made the modification manually, but this workaround can be done by tuning the travel speed in advanced settings in Cura 
 
Just let me know

64Florian, 2020-04-29:
hmm @Ghostkeeper ... sorry to boring you, but:  
According this page : [marlinfw.org ->G0-G1](https://marlinfw.org/docs/gcode/G000-G001.html)
"_By convention, most G-code generators use G0 for non-extrusion movements (those without the E axis) and G1 for moves that include extrusion._"
means if I correct the page : 
G1 [E < pos >] [F < rate >] [X < pos >] [Y < pos >] [Z < pos >] 

so according my last message, 

> **G1 F1200 X54.468 Y110.946 E0.00916** _-> go to X54.468 Y110.946 E0.00916 @ 1200mm/min (20mm/s) (Linear Move)_

is **bad** due to position of Feed rate and Extruder !

and still according this page, due to nothing change the feed rate for X,Y after the "_G0 F6000 X54.148 Y111.073 Z0.5_ " this will be the speed of the print according @iammenotyou issue
=> that's why my sample should work

Ghostkeeper, 2020-05-04:
> hmm @Ghostkeeper ... sorry to boring you, but:
> According this page : [marlinfw.org ->G0-G1](https://marlinfw.org/docs/gcode/G000-G001.html)
> "_By convention, most G-code generators use G0 for [...]

In g-code the order of the parameters doesn't matter. For example on that same page they also give the following example g-code: `G1 X50 Y25.3 E22.4 F3000` where the parameters are ordered differently as well.

64Florian, 2020-05-04:
Thank for this information Ghostkeeper.
so waiting for @iammenotyou feedback with my file ... (should not improve if position doesn't mater in his printer too)

ilko-k, 2020-05-06:
hello,
I'm experiencing a similar issue since a while and finally decided to report it. The wall speed never matches the initial layer speed and is printed at full regular speed. As long as I remember, this issue wasn't occurring in Cura v3.6.

[image]

as you can see in the attached screenshot, the skirt is set to 16 mms like the initial speed itself. But when the print starts, the walls are printed at 60/80 mms. It seems like when the wall speed is set manually, the initial speed is ignored.

64Florian, 2020-05-06:
Hello @ilko-k  
- Could you try to set manually initial travel speed at 20 mm/s and "listen" if 1st layer is printed at the same speed as the travel speed... 
- Did you try to switch to auto all settings to say:
> when the wall speed is set manually, the initial speed is ignored
- Do you know the marline version used by your printer ? 
- In the preview, could you have a look for speed "colors" like that: 
> [image]
- finally, could you share a sample gcode, for @iammenotyou I was not able to see any "bad" speed, but it could be in your case ...
I add a basic 1cm cube for easier gcode analysis:
[1cmCube.zip](https://github.com/Ultimaker/Cura/files/4589656/1cmCube.zip)
It will be nice if you are able to share the gcode and try to record the "sound" for your printer from the home init to the printing of the 2nd layer (as MP3 or wave file)

Thx

Ghostkeeper, 2020-05-07:
Like I mentioned before, to me a project file would be very useful to debug your issues. You can make one by reproducing the issue in Cura and then clicking File -> Save. Share the file that is produced then here (you can rename it to .zip or something to upload to Github).

This project file includes your build plate as well as all the profiles and settings you are using.

ilko-k, 2020-05-07:
@64Florian @Ghostkeeper Just to be clear, I know how to analyse a gcode file. Whatever the settings I am specifying, it has no effect on the initial speed. That said, feedrate color scheme is working correctly and indicates the expected speed.

My device is a Tevo Flash running Marlin 2.0.5.3.

I've attached an archive with different settings as demanded.
[sample_cube.zip](https://github.com/Ultimaker/Cura/files/4596139/sample_cube.zip)

64Florian, 2020-05-08:
@ilko-k, I don't know your skill with GCode ... (How could I ? you didn't said you made GCode comparison to see the speed of the first layer - as you can read, _iammenotyou_ have not your skill and tune his printer on each print instead of editing the gcode ...  but  it's not a proper way to work with this tool for sure)
I try to help because I like this software and I want to improve it ... unfortunately I'm not able to compile the code right now, the gap it's too big for me currently not yet fluent with Git...
Many times in software support, the issue isn't in the code but the way the user use it. That's why starting from basics can prevent spending hours on the code for dev teams to try to fix nothing ...

As I can see in your GCode you are in 4.4.1 version ...
I have opened a ticket and made a analyses for profile issue mismatch:
_profiles settings not matching up in the application_ #7626
-> in fact since 4.4.0 beta-pre-release a new profile management was created and a big regression seem to be occurred since this date with matching profiles and settings 
In another word, have a look in your profile folder for extruder AND printer settings, you could have the same issue as me : double speed system = What is displayed isn't what is used !
another person have this kind of problem : _Infill Speed bug_ #7702

_Ghostkeeper_ need a saved workspace (as it ask since a long time) to be able to "reproduce" on his computer the issue (it can made code analysis with that, and it's the only solution to do that  because if you take a "fresh install" with clean new profiles and try to reproduce you will not have any issue)
Have a look on my 4.6.1:
[image]
I try on 4.4.0 Beta too:
[image]

=> means not really a "software issue" but more a less a migration issue ...

ilko-k, 2020-05-08:
Hey, I did investigate a bit more and figured it out. I was using a modified definition file for the profile creation of my device and named it "tevo_flash.def.json". Here is the part that what was causing the conflict : 

`tevo_tornado_extruder_0`

My device is a Tevo Flash and still not supported by Cura so... I created a custom .def.json file for it but I had to use an existing definition for the extruder otherwise Cura was crashing. Creating a profile using Cura's normal process is the way to go.

closed 2020-06-19 by no-response[bot]