Ultimaker/Cura #7521: Pause at height inserts into wrong location

kejones62, 2020-04-17:
**Application version**
4.5.0

**Platform**
Windows 10

**Printer**
Creality Ender 3

**Reproduction steps**
1. Opened "Guide_Open.STL" file
2. Opened "End_Clip.STL" file
3. Used 'Pause at Height' to insert a pause at 5mm.   All default settings for this script were used.
4. Sliced and printed gcode
5. After getting a bad print, I examined the gcode.  The pause seems to be inserted into the middle of the 4.8mm layer.  As a result, the code does a portion of the  4.8mm gcode, pauses, goes to 5mm height as per script, then resumes the 4.8mm gcode but at a 5mm height.  It then encounters the 5mm layer, and executes this, also at 5mm.
 
**Screenshot(s)**
gcode after insertion of pause:
G0 F300 X134.53 Y139.269 Z4.8      <------ Working on layer 4.8
G0 F9000 X133.864 Y143.136
G0 X132.785 Y143.136
G0 X109.197 Y156.874
G0 X108.114 Y156.916
G0 X106.153 Y163.751
;TIME_ELAPSED:3175.998973   
;TYPE:CUSTOM
;added code by post processing      <------- Not done with 4.8mm stuff, but starting pause!
;script: PauseAtHeight.py
;current z: 5
;current height: 5.0
M83 ; switch to relative E values for any needed retraction
G1 F300 Z6 ; move up a millimeter to get out of the way
G1 F9000 X190 Y190
G1 F300 Z15 ; too close to bed--move to at least 15mm
M104 S0 ; standby temperature
M0 ; Do the actual pause
M109 S200 ; resume temperature
G1 F300 Z6
G1 F9000 X106.071 Y162.339
G1 F300 Z5 ; move back down to resume height    <-------- Goes to 5mm
G1 F2700 ; restore extrusion feedrate
M82 ; switch back to absolute E values
G92 E2062.81891
;LAYER:23
;TYPE:WALL-INNER
;MESH:Guide_Open.STL
G1 F2700 E2067.81891       <------- starting doing stuff that should have been done at 4.8mm
G1 F1500 X106.075 Y162.408 E2067.86365
.
.
[ lots of gcode skipped]
.
.
;MESH:NONMESH
G0 F300 X134.53 Y139.416 Z5   <--------- Here's the start of the 5mm layer!
G0 F9000 X133.864 Y143.136
G0 X132.785 Y143.136
G0 X109.2 Y156.963
G0 X108.118 Y157.006
G0 X106.153 Y163.751
;TIME_ELAPSED:3296.484205
;LAYER:24
;TYPE:WALL-INNER
;MESH:Guide_Open.STL
G1 F2700 E2138.78564

**Actual results**
See above

**Expected results**
Postprocessing  should insert the pause at the end of the layer preceding the pause height - not in the middle

**Project file**
[CE3_Guide_Open_test_pause.zip](https://github.com/Ultimaker/Cura/files/4490756/CE3_Guide_Open_test_pause.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
Couldn't find a log file!

**Additional information**
Brand new to 3DP, Cura, and gcode so I hope this isn't bogus.  Searched and didn't see any identical open bugs.  Be gentle.

Ghostkeeper, 2020-04-22:
> Brand new to 3DP, Cura, and gcode so I hope this isn't bogus. Searched and didn't see any identical open bugs. Be gentle.

It's not bogus. It's a pretty good find, sharp eye. However this issue is indeed already reported to us before, albeit not with a very descriptive title. Please see https://github.com/Ultimaker/Cura/issues/5085. I'll adjust the title of that issue. It's on our backlog. We see the value in that bug being fixed.

closed 2020-04-22 by Ghostkeeper

kejones62, 2020-04-22:
Thanks.  I saw 5085 but it didn't seem the same.  Just to be clear, what I
see is not just "at the wrong layer/height", but also "in the middle of a
layer/height".

Ghostkeeper, 2020-04-22:
It's inserted right before the `;LAYER` comment. There's just a lot to insert :smile: