Ultimaker/Cura #7522: [4.5.0] brim not printed in one of a two object stl file.

Fedonecchia, 2020-04-17:
**Application version**
4.5.0
**Platform**
macOS Catalina (10.15.2)

**Printer**
Anet A8

**Reproduction steps**
1 imported an stl file consisting of two objects spaced out by a few millimiters.
2 sliced and exported gcode
3 looked at printer
4 looked at preview's first layer

**Screenshot(s)**
[image]

[image]

**Actual results**
spaghetti
[image]

**Expected results**
prints brim on both structures

**Project file**
[simple box v0.stl.zip](https://github.com/Ultimaker/Cura/files/4492523/simple.box.v0.stl.zip)

**Additional information**
I had tree supports on, only on the buildplate. Don't know if that's relevant.
Also, when I imported the same objects as separate stl files the brim was there and the print went fine.

smartavionics, 2020-04-17:
it's because when you rotate the model to that orientation what are now the bottom surfaces are not at the same height. You have to sink the model into the buildplate to ensure that both bottom surfaces intersect with the buildplate.

[image]

Fedonecchia, 2020-04-17:
Thanks a lot! do I need to mark the issue as closed or something?
fedo

closed 2020-04-18 by Fedonecchia