Ultimaker/Cura #7523: Adjustable coordinate precisino for generated gcode

BarsMonster, 2020-04-17:
When printing smooth curves from tiny segments in Marlin - it was identified that estimated print speed in Junction Deviation mode is very sensitive to junction angle between 2 consecutive segments. It is currently seen that angular error of just 0.1° could significantly affect print speed and hence - ruin surface quality. Some fixes will be done in Marlin, but I can see that 1um resolution that Cura is currently using for output is not enough to calculate angle with sufficient precision for perfect surface quality. 

**Describe the solution you'd like**
Add settings to change default resolution of coordinates in generated gcode for XY, Z and E. Default is 3 for both XY and Z, and allow to increase it to 4 or 5.

**Describe alternatives you've considered**
The only alternative is to print with larger segments, which will reduce surface quality of smooth segments. 

**Affected users and/or printers**
Default settings exactly match current behavior, so no impact on existing user base.

Ghostkeeper, 2020-04-22:
Interesting discussion. Also, long discussion :smile: Funny how people independently arrive at the same problems in completely different domains.

To adjust the precision we'd need to adjust the `INT2MM` and `MM2INT` and related macros in `utils/Coord_t.h`, and also the `writeInt2mm` function family in the g-code export. However sprinkled throughout the code base there are also a few constants, which probably shouldn't be there in the first place but are, like maximum distances at which to merge lines together or something. Those would have to be weeded out. Not all constant `coord_t` types are like this though, since most of them are fudge factors to deal with accumulated rounding of coordinates, which can be left as-is (usually 10 units or so).

Also note that the maximum printer size would be decreased. We currently use signed 64-bit numbers for our coordinates, meaning that we can handle up to 2.1km in theory (way bigger than necessary really) since we need to calculate distances using Pythagoras which involves squaring these numbers and adding two of them together. However in reality we're probably somewhat more limited because we also do things like sum the total circumference of contours, and the total length of combing paths. It's probably safer to assume that the limit is currently more like 500m. If we were to increase precision to nm this would become 0.5mm then, which is more in the range of the large-ish home use printers on the market now.

Sophist-UK, 2020-04-24:
Marlin supports circles, arcs and Bézier cubic splines so **if** it were possible to use these, you would likely get much smoother walls.

Of course, STL files (and possible other model file types) consist of triangular patches with straight edges, and working out that a series of short straight G1 moves are actually part of one of these curves and replacing them with an actual curve may be at best non-trivial in compute time or even impossible. Additionally you would have the issue of ensuring that the same curve is used for the same segments on adjacent layers - so probably impossible to achieve in practice. But a nice thought.

Ghostkeeper, 2020-04-24:
We've tried arcs and splines before, internally at Ultimaker, but it had two problems, one of which you mentioned:
* Generally the implementation in Marlin was too naive to make this practical. As I understand it, the Marlin implementation would discretise the curve to segments which became longer and coarser if the curve had a bigger scale, which kind of defeats the purpose. I understand now that the Marlin implementation has improved since then, but since Ultimaker's printers have forked off Marlin some 6 years ago or so, we won't be able to join in that bounty.
* The slicer would need to fit curves on the surface. When faced with a collection of triangles that approximate a curve, it's ambiguous whether the user meant to create a curve there or actually wants those corners. You can have settings that try to adjust the parameters and sensitivity of this matching but it will be quite hard to get it right for all models or even for most.

Ghostkeeper, 2020-04-24:
I brought this up to our team but we've decided to put off this issue for later. The microstepping range of our printers (and most printers currently on the market) is already way bigger than a micrometer, and play in the gantry is even more of an effect. When using the "classic jerk", the shallow 0.1° corners will not reduce the speed of the printer at all. For instance with a jerk of 10mm/s you would start getting a slowdown from [5.73m/s and up](https://www.wolframalpha.com/input/?i=10+%2F+sin%280.1+degrees%29), which is way beyond the range of any current printer.

I don't know how the new methods in Marlin cope with this though. For Ultimaker this is irrelevant since we are not using those. However that means that we can't justify planning this in until our own printers have any benefit from this feature.

BarsMonster, 2020-04-24:
@Ghostkeeper Ok, that is clear and understood. 

If there will be anyone stumbling onto this over the following years - I can suggest to artificially increase size of printer by a factor of 10-100 with appropriate change of layer height and extrusion multiplier + gcode postprocessing to scale back coordinates or change steps per mm in your printer. 

That's a dirty hack, but it works for initial tests.

BarsMonster, 2021-03-23:
Now with "arc welder" plugin widely used, and significant space savings offered by arc welding - I think this option has more sense to exist, even though current arc welder plugin allows for 0.03mm deviations.