Ultimaker/Cura #7524: [4.5.0] Wall Flow Settings change all G-Code

filipesbp, 2020-04-17:
**Application version**
4.5.0
**Platform**
Windows 10

**Printer**
Ender 3

**Reproduction steps**
With a project open, example calibration cube and Settings = Expert
1. Confirm Inner Wall Flow is 100% 
2. Saved G-Code to file: Original
3. Changed only the Inner Wall Flow to 105% 
4. Saved G-Code to file: Flow105
5. Compared the two files in notepad++

**Expected results**
Expected only the Extrude value change on the Inner Wall.

**Actual results**
All Extrude values is changed:

Original file:
;TYPE:WALL-INNER
G1 F2700 E0
G1 F1200 X108.9 Y108.9 E0.61876
...
;TYPE:WALL-OUTER
G1 F1200 X107.7 Y107.7 E8.38917

Flow105 file:
;TYPE:WALL-INNER
G1 F2700 E0
G1 F1200 X108.9 Y108.9 E0.65445
...
;TYPE:WALL-OUTER
G1 F1200 X107.7 Y107.7 E8.83249

**Additional information**
Tested on beta version 4.6.0 with the same result.

**Conclusion**
When switching the Inner Wall Flow all flow rates is changed too.

smartavionics, 2020-04-17:
Of course, one factor in this is whether you are using relative or absolute extrusion. If absolute, you would expect any change in extruded length on one line to affect all the E values on the following lines up to the next extruder reset. If you are using relative extrusion, you wouldn't expect a change in extruded length on one line to affect the following lines.

filipesbp, 2020-04-17:
You're right, the inner wall is the first thing that prints. I've tested changing only the outer wall and the first part of the g-code its the same, after the inner wall all extrude values are different.

closed 2020-04-17 by filipesbp