Ultimaker/Cura #7525: Outer wall wipe distance, not on first layer?

ashleysommer, 2020-04-18:
** Description of the problem **
I like to use the "outer wall wipe distance" feature to weld my outer wall seams. This feature works well for me, except I've found it sometimes causes problems on the initial layer. When the hotend nozzle moves back over a section of extruded outer wall to weld it, that can remelt that small section just enough to cause that area to lift off the build surface. This causes a small blob in our outer wall on my initial layer.

**Describe the solution you'd like**
I would like the ability to have zero "outer wall wipe distance" for the initial layer, then revert back to normal setting for subsequent layers.

**Describe alternatives you've considered**
Simply disabling "outer wall wipe distance" for the whole print works to eliminate this problem, but it means I can't use this feature for the rest of the print. I haven't tried any other slicers, I prefer to use Cura.

**Affected users and/or printers**
I use an Anycubic Mega S. I don't know if this affects any other printers.

**Additional context**
Note this might be affected by my use of 115% flow rate on the initial layer, perhaps "outer wall wipe distance" is not compatible if initial layer flow rate is > 100%?

Ghostkeeper, 2020-04-24:
We've discussed this with our colleagues but we think that the change will not bring about a great improvement. Generally the wipe distance is so small that it doesn't really have an effect and could even re-melt it and overextrude on the seam, which adds to the adhesion.

We're not entirely against it though, but we'll not give this any priority. It's deferred until priorities change.

ashleysommer, 2020-04-24:
@Ghostkeeper 
Thanks for your response,
I've printed a new parts cooling duct, and slightly increased my first-layer fan speed, and seem to have solved my problem for now anyway.

closed 2023-07-13 by github-actions[bot]