Ultimaker/Cura #7527: [4.6] Print profiles not listed in dropdown menu but still exist in manage profiles

misslittlem, 2020-04-19:
**Application version**
(The version of the application this issue occurs with.)
4.6 beta
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
windows 64
**Printer**
(Which printer was selected in Cura?)
ultimaker 2 w olsson block and upgraded ddg extruder
**Reproduction steps**
1. (Something you did.)
looked for profiles in the dropdown menu
2. (Something you did next.)
found the print profiles in the manage profiles and was able to switch between them. (this has not been possible for the past 2 versions of cura, 4.4.1 always crashed and i was never able to use it. 4.5 i was able to open and see the print profiles in the manager but not in the dropdown however cura crashed everytime i tried to 'activate' a pre existing profile.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)
list of profiles not visible in the dropdown menu
**Expected results**
(What should happen after the above steps have been followed.)
should have the profiles visible and selectable without having to go to the manager.
[image]

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-04-22:
Hmm, I'm unable to reproduce that problem. For me it just shows the four available quality profiles:
[image]
I don't know what you mean by that "ddg extruder" though. Is that an upgrade you applied in Cura as well or is it just on your printer?

Maybe it would help if you were to save a project. You can leave the build plate empty, but project files contain your full configuration so we may be able to see what is going wrong then.

uyesercan, 2020-04-22:
I have same bug with 4.6 (not beta). My print profiles not listed in dropdown. 4.5 its ok but 4.6 not show print profiles with same PC.

[image]

uyesercan, 2020-04-22:
Btw in my case, i discovered it looks like not a bug. My situation it's about printer profile.

If i select Creality Ender-3 **Pro** not show any print profile. If i select Creality Ender-3, show profiles like old (4.5) version. So, the Creality Ender-3 **Pro** version does not contain a printer profile ...

fieldOfView, 2020-04-23:
@uyesercan, see https://github.com/Ultimaker/Cura/issues/7550

whoisak, 2020-04-28:
> Btw in my case, i discovered it looks like not a bug. My situation it's about printer profile.
>
> If i select Creality Ender-3 **Pro** not show any print profile. If i select Creality Ender-3, show [...]

This has been my experience as well. after upgrading to 4.6, using the new Ender 3 Pro printer template to create a printer, the existing profiles i have do not show up. 

Also, looking at the existing materials I have entered, the Ender 3 Pro printer, when active, will not allow selection of a nozzle width
[image]
[image]

raf802, 2020-04-28:
> This has been my experience as well. after upgrading to 4.6, using the new Ender 3 Pro printer template to create a printer, the existing profiles i have do not show up.
>
> Also, looking at the [...]

I too have this issue with my Ender 3 Pro when I use its template.

mbaldwin0352, 2020-05-09:
Cura 4.6.1 missing drop downs as well but I have 2 printers.

I activated the other printer and the profiles returned to the menu drop down. I switched back to my 1st printer and the profiles are back.

The only thing I didn't do was save edits to the profile I was changing last night before shutting down my computer.

Ghostkeeper, 2020-05-12:
If you've added an Ender 3 Pro in 4.6.0 that didn't get any nozzle, you'll need to select a nozzle from the available nozzles in 4.6.1 before you get any quality profiles.

mbaldwin0352, 2020-05-12:
all printers pre-existed and were working normally. in my case the same printers have been installed since 3.8 yet I am still encountering the issue daily after 4.6.1 update

Ghostkeeper, 2020-05-14:
That's funny, because Cura 3.8 doesn't exist! We went from 3.6 to 4.0. Which printers are you talking about?

mbaldwin0352, 2020-05-14:
your are correct - sorry for the confusion.

the issue only is occurring in 4.6.1  for me.

below  is the image of the profiles being blank.

steps:
reboot computer
launch cura 4.6.1
Change to different printer
Profiles are blank

keep in mind I did not check the profiles with the printer profile that was currently loaded. I only launched the app and then switched the printer.
[image]

mbaldwin0352, 2020-05-14:
in addition noticed that if i go from a custom profile to a built in profile and then back to a custom profile the stock profiles are gone:. scrolling up bounces the drop down but only show the custom profiles.

[image]

mbaldwin0352, 2020-05-14:
reinstalled cura 4.6.1 and the issue remains

Delete all printers from Cura 4.6.1

Added two Ender 3 printers (not the pro profile)

Imported gcode from original printers (copy and paste into start gcode text )

ALL profiles visible on first printer. custom and built in

Changed to second printer and ALL profiles are visible.

Rebooted computer and launched 4.6.1 

duplicated above and all profiles are back

had to delete ALL printers to get them to return - I am using it throughout the day and will record steps prior if the issue returns. I wouldn't close this yet (if possible) because the issue still remains. had to deleted printer profiles. Since there isn't a clear way to backup the files within cura without exporting an entire project ( not a named feature of export machine. this is quite an annoyance and interruption of work flow for our business.

Ghostkeeper, 2020-05-18:
Right, so probably it was something in those specific printers that are now deleted. We'll have to keep an eye out if we can see the issue return.

Neofire001, 2020-07-11:
I am having the same issue and I was able to track it down to a custom "Material Type" name.  I added a custom material type of "PLA Silk".  I selected the new material and the dropdown listed all the profiles as expected.  Once I restarted the software the dropdown menu no longer listed any profiles.  However, they are still in "Manage Profiles".  Once I renamed the material type to PLA and re-selected the material for printing the list re-populated immediately.  The dropdown list also persisted after software restarts.

System Specs
Windows 10 x64
Cura 4.6.1 
Ender 3 Pro

Ghostkeeper, 2020-07-14:
Indeed we actually got this reported through Ultimaker's support channel recently as well. I'll reopen this. There are several issues with changing the material type of a printer:

* The active quality and intent profiles are not changed, even though the list of available profiles changes.
* New custom profiles don't always show up in the profile list.
* The list of available profiles in the profile manager isn't updated properly.