Ultimaker/Cura #7528: Please add an option to orbit camera around a selected model

brolifen, 2020-04-19:
When you're dealing with multiple items on the printbed and you want to inspect them one by one it gets very tedious because the camera always orbits around the center of the build plate by default. This makes it very difficult to hone in on a model to inspect it. 

Either an option to set the new camera orbit center to the current selection or an option to "isolate" the current selection until you exit out of this isolation mode and make it the new center of orbit could solve this problem.

ScruffR, 2020-04-19:
This would be a good idea - especially since it also appears that "Zoom toward mouse direction" doesn't seem to working anymore either - or at least I cannot see any difference in behaviour whether this option is on or off (I'm not using a mouse but touch pad and/or touch screen).

nallath, 2020-04-20:
Preferences -> Center camera when item is selected.

ScruffR, 2020-04-20:
Nope, that is not exactly the solution we are looking for.
While this option does somewhat provide that experiece it will only ever rotate around the center of the screen, but sometimes it would be preferable to change the location of the pivot point off screen-center.

The existing feature places the _object_ at the orbit center point, the proposed feature would put the _orbit center_ *at* the position of the object center.

And the "Zoom toward mouse direction" is also not addressed by just closing the issue.

nallath, 2020-04-20:
We will only look at a single issue per ticket, it's impossible for us to keep track of things otherwise. So if there are multiple bugs/features, please post _multiple_ issues.

smartavionics, 2020-04-20:
Actually, I think you will find that it rotates around the "origin" which, to begin with, is located in the centre of the buildplate but when you scroll around using shift-left button, it moves the origin and so you can place the centre of rotation wherever you wish.

ScruffR, 2020-04-20:
Thanks @smartavionics, I'll have to try that.

Ghostkeeper, 2020-04-22:
> an option to set the new camera orbit center to the current selection

I mean, that sound pretty much like "Center camera when item is selected" to me :smile: I pretty much can't live without that option. I'm so used to it. It's the first thing I enable every time I wipe my preferences (a kinda regular occurrence as a developer).

Why would you want to rotate around a point that's not in the centre of your view? That could become *very* confusing very quickly. For instance when you're rotating around something on the side of your view, then zooming in, then now you're rotating around something that's behind you. It'll be hard to get out of that situation if you handle it wrongly.

I don't really see the use case for pivoting around a point that's not in the centre of your view. I also don't know of any software that does this, aside from a few games where the character would be in the way of the player's view if it were in the middle. I think the Witcher had it. But that's not really a use case for Cura.

ScruffR, 2020-04-22:
> I don't really see the use case for pivoting around a point that's not in the centre of your view. 

Fusion 360 and Blender do.

You can set the orbit point wherever you want and rotate your world around it - and for me, that is something I'd be/am missing when it's not there 😉

Ghostkeeper, 2020-04-28:
I may be misunderstanding you then. I've used Blender a lot (although it still has lots of features that I've never discovered) but I don't know of a way to make it rotate the viewport around a spot that's not in the centre of the viewport. I know that you can make objects rotate around specific spots (like with the . period hotkey) including the camera object from where the render is made, but not the Blender viewport itself for the modeller.

I also don't really see the point for Cura. Most of the time you want to either oversee the entire build plate, or zoom in on a specific piece of layer view. You don't need to make the camera make smooth curves through your scene.

closed 2020-06-19 by no-response[bot]