Ultimaker/Cura #7529: Dual extrusion does nozzle switch heat-up ontop of object

ScruffR, 2020-04-19:
Is this intentional? I wouldn't see why it is doing it, but during the heat-up phase of the next to use nozzle, the nozzle head is parked over the object instead of outside the ooze shield.
This doesn't make sense to me. I would have expected that this exactly is what the ooze shield is meant for, that any oozing during the heat-up phase would be wiped off on the *outside* of the ooze shield to *not* end up *on* the object.

Ghostkeeper, 2020-04-22:
This depends on your printer's settings. It's configurable via the [machine_extruder_end_pos_abs, machine_extruder_end_pos_x and machine_extruder_end_pos_y](https://github.com/Ultimaker/Cura/blob/a0dc78cb391272d69f3d16e4954ce8854a60fd89/resources/definitions/fdmextruder.def.json#L135-L169) settings. These specify where it should start the extruder switch. For instance, the Ultimaker 3 puts this around the location of the switching bay on the side of the printer. Then the `machine_extruder_start_pos_xxx` settings specify the location where the extruder starts off from after it's completed switching. This may be different because the act of switching may move the head, and the other nozzle may have a different position on the print head.

By default these are set to be not absolute (so relative to the last location of the nozzle before switching) and to the 0,0 coordinate. That basically means that it's going to switch at the location of your last print. This is the safest because moving in any direction could ram the print head into the side of the printer which can damage the printer, and this is the global default which has no information on which location of the printer would be a good place to move to then.

closed 2020-04-22 by Ghostkeeper

ScruffR, 2020-04-23:
Hmm, too bad. 
I hoped Cura could assist here as _it_ knows the shortest distance to the ooze shield and also knows the nozzle offset of either nozzle on the shared "head" (although missing an option to explicitly state it is shared - there is a shared nozzle setting tho').
Given this info Cura should be able to know how to place both nozzles just outside the ooze shield - at least for a multi-nozzle/shared head setup.

For independent heads an absolute parking position makes sense but for a shared head this would mean extra travel for nothing.

Ghostkeeper, 2020-04-28:
The ooze shield is not always enabled. It gets enabled in less than 0.1% of the prints as far as we can see. It usually won't necessarily be a good spot to move towards.

The nozzle offset or whether there is a shared nozzle or not doesn't necessarily matter for the `machine_extruder_end_pos_*` settings. For some printers the act of switching itself also causes the print head to move, like for instance with the Ultimaker 3 which has a switching bay and a lever that the print head will toggle by moving in a curve around the switching bay. That movement towards the switching bay and around the switching bay causes the print head to end up in a wholly different position as where it started. With `machine_extruder_end_pos_*` we can capture all of those cases, except cases that depend on the model's shape.

> For independent heads an absolute parking position makes sense but for a shared head this would mean extra travel for nothing.

So this is where the `machine_extruder_end_pos_abs` setting comes in. It doesn't need to be an absolute position. By default it's a relative position.

ScruffR, 2020-07-01:
Sorry for the late reply, but thank you for the response.
I'll try to find my way round with this info :+1: