Ultimaker/Cura #7530: [4.5.0] Profiles section in Settings window doesn't show differences to active profile anymore

Triangulix, 2020-04-19:
< !--
Processing an issue will go much faster when this is filled out, and issues which do not use this template WILL BE REMOVED and no fix will be considered!
-->
I love you, too.

**Application version**
4.5.0

**Platform**
Windows 10, Nvidia GeForce GTX 1050

**Printer**
Custom Profile

**Reproduction steps**
1. Select some profile in the drop down
2. Open the drop-down again and select Manage Profiles...
3. select another profile in the now open window

**Screenshot(s)**
[image]

**Actual results**
The Extruder tab is empty.

**Expected results**
I expect to see the differences of the selected profile compared to the active profile as it was in Cura 3.6.0

**Additional information**
I just switched from Cura 3.6.0 to Cura 4.5.0.

Ghostkeeper, 2020-04-22:
Yeah, I could reproduce this issue. It seems to be specific to per-extruder settings.

Vandrasc, 2020-04-23:
Devs see CURA-7392 for reference.

commit 392c4cfe (2021-06-10): Fix extruder profiles not being shown in profile preferences

CURA-7392

Fixes #7530
closed 2021-06-10 by Ghostkeeper