Ultimaker/Cura #7531: "Arrange all models," please consider layer-count

FlailAway, 2020-04-19:
**Is your feature request related to a problem? Please describe.**
Arrange all models sequence selections seems to be based only on area.

**Describe the solution you'd like**
Arrange the models on decreasing (or increasing) layer heights and/or area.

I have 3 items on the bed and I clicked "Arrange all models" (print all at once) and without thinking about it, sent it all to the printer. But now the printer is scooting back and forth skipping over the large area (4" x 6") middle item as it was only 13 layers high. The other items are many more layers but it is wasting a lot of time skipping over the middle part at every layer increase.

Ghostkeeper, 2020-04-22:
This algorithm is due some rework, but I don't think this is taken into consideration yet with our new requirements. The reason for that is that this sort of optimisation is often much smaller than what most people seem to think. Take this pretty extreme example:

[image]

[image]

We're saving on a grand total of 0.2% print time with that change, and part of that is due to having a shorter skirt.

Sophist-UK, 2020-04-24:
@Ghostkeeper - If you are reworking the auto placing, then can you please have different algorithms for All-At-Once and One-at-a-Time with One-at-a-Time placing them in sequence front to back (or back to front - see below) starting with the object with the smallest Y-size.

The choice of front-to-back or back-to-front depends on the print head dimensions, specifically the Y-min / Y-max settings. If Y-max is **less than** Y-min, then it should be front-to-back; Y-min less than Y-max, back-to-front.

Additionally, I would also suggest positioning in a fashion that maximises distance from the home position i.e. assuming home is rear-left, then objects should be positioned front-left to rear-right. 

The purpose of this is to align this with a PR I hope to submit which changes how Cura deals with models whose height is above the gantry height, allowing multiple objects over gantry height to be printed providing that the gantry and head can pass safely behind the next model.

Ghostkeeper, 2020-04-28:
> The choice of front-to-back or back-to-front depends on the print head dimensions, specifically the Y-min / Y-max settings. If Y-max is less than Y-min, then it should be front-to-back; Y-min less than Y-max, back-to-front.

It's a little bit more complicated than that. In the user interface we've only exposed a customisation for rectangular print heads, but Cura models the print head as an convex polygon and can resolve the order that way. At least it could, before this got broken in 4.6 a bit again.

Sophist-UK, 2020-04-28:
Yes - from starting to look at the code I am aware of that.