Ultimaker/Cura #7532: Cura 4.5 slicing off center

Silversurfer79, 2020-04-19:
Can any tell me why Cura 4.5 is doing this and how to resolve this issue please? 
[image]

If I slice with Cura 4.4.1 its perfect.
[image]

It seems that the start G Code has changed, and which U have update and has made now difference. 
[image]

fieldOfView, 2020-04-19:
It looks like you have selected the “Origin at Center” setting in the Machine Settings, which should only be selected for delta-style printers.

Silversurfer79, 2020-04-19:
> 
> 
> It looks like you have selected the “Origin at Center” setting in the Machine Settings, which should only be selected for delta-style printers.

Thats it thanks! Have a good reset of your day ;-) Stoked!

closed 2020-04-20 by nallath