Ultimaker/Cura #7533: Cura 4.5 Small portions of STL object sliced as empty space

Shadrach901, 2020-04-19:
[image]
[image]
[image]

Cura version 4.5. windows 64bit. has an ssd and upwards of 2.5ghz GPU

Slicing for a CR-10S, however, the issue iteration pictured was after re-downloading the latest version of the software without loading in a printer profile.

Originally printed on an older version and got a much more significant gap in the same area(about 1/8 inch tall).
After the print came out i troubleshot the older version. tried rotating the object, re-
scaling, changing layer heights. all these options changed the thickness of the height of the gap, but the gap stayed relatively similar in position relative to the build plate. 

so i downloaded the new version of cura, and the gap is much smaller but still exists. went through same troubleshooting steps, and results are the same. pictures are from the newest version of cura. first in solid view, to show the integrity of the STL as loaded by cura in that area, then in layer view to show the issue after sliced. and then rotated 15degrees to show that the issue remains parallel with the build plate.

for reference, I also downloaded prusa and sliced the same stl with no issues. so it does not seem to be an stl issue.

the fact that scaling and rotation affect the size of the area, but not its oriontation to the build plate, leads me to believe it is a coding issue outside of my ability to fix.

log
[cura.log](https://github.com/Ultimaker/Cura/files/4500173/cura.log)

3mf saver file
https://we.tl/t-MonnqenWRF

I have had this issue once before when scaling up a very small STL nearly 1500%, so i assumed it was a mesh integrity issue.

I looked through the available bug reports and didn't see anything like this. if the issue already exists, let me know and i will refer to that thread.

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues
 to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

smartavionics, 2020-04-19:
The model has errors. If you install the Cura mesh tools plugin you will see the warning when you load a bad model...

[image]

The purple/blue regions are displayed by a new cura feature that highlights bad areas of the model. That feature hasn't yet been released but the mesh tools are available.

closed 2020-04-20 by nallath

Shadrach901, 2020-04-22:
thanks for the help.
 is this the tool you're referring to?

https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

smartavionics, 2020-04-22:
Yes.