Ultimaker/Cura #7535: Cura 4.5 Wont load older Profiles and app crashes when attempting to delete old profiles

etb1989, 2020-04-20:
Processing an issue will go much faster when this is filled out, and issues which do not use this template WILL BE REMOVED and no fix will be considered!

Before filing, PLEASE check if the issue already exists (either open or closed) by using the search bar on the issues page. If it does, comment there. Even if it's closed, we can reopen it based on your comment.

Also, please note the application version in the title of the issue. For example: "[3.2.1] Cannot connect to 3rd-party printer". Please do NOT write things like "Request:" or "[BUG]" in the title; this is what labels are for.

Thank you for using Cura!
-->

**Application version**
4.5

**Platform**
Window 10 64 

**Printer**
Tevo Tornado

**Reproduction steps**
1. Updated to 4.5
2. attempted to use older profiles

**Screenshot(s)**
Nope

**Actual results**
The app crashes when older profiles are deleted

**Expected results**
profiles to work or be deleted

**Project file**
no

**Log file**
no
**Additional information**
no)

nallath, 2020-04-20:
If you only fill in the template partially, there is a chance that we don't have enough information to debug the issue. This is the case right now.

Could you please share your configuration folder and logs?

etb1989, 2020-05-01:
Hello Github, Could you please tell m how to do this please. Regards Elliot  Sent from Mail for Windows 10 From: Jaime van KesselSent: Monday, 20 April 2020 6:50 PMTo: Ultimaker/CuraCc: etb1989; AuthorSubject: Re: [Ultimaker/Cura] Cura 4.5 Wont load older Profiles and app crashes when attempting to delete old profiles (#7535) If you only fill in the template partially, there is a chance that we don't have enough information to debug the issue. This is the case right now.Could you please share your configuration folder and logs?

	
        
		Virus-free. www.avast.com

mahtDFR, 2020-05-01:
I removed some of the info from your previous comment for readability.

We need you to completely fill in the issue template, otherwise we don't have any useful information to help you fix your issue, and that's sad.

I've pasted it below. Please, please, please fill it in entirely, and then we will do our best to help you.

> 
> 
> **Application version**
> (The version of the application this issue occurs with.)
> 
> **Platform**
> (Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
> 
> **Printer**
> (Which printer was selected in Cura?)
> 
> **Reproduction steps**
> 1. (Something you did.)
> 2. (Something you did next.)
> 
> **Screenshot(s)**
> (Image showing the problem, perhaps before/after images.) 
> 
> **Actual results**
> (What happens after the above steps have been followed.)
> 
> **Expected results**
> (What should happen after the above steps have been followed.)
> 
> **Project file**
> (For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
> 
> **Log file**
> (See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
> 
> **Additional information**
> (Extra information relevant to the issue.)
>

etb1989, 2020-05-05:
*Hello GITHUB,*

*Below are my answers to your questions.*

*Application version*
Cura 4.5

*Platform*
Windows 10 on intel i5-3317

*Printer*
Tevo tornado

*Reproduction steps*

   1. Attempted to use profiles from a previous version
   2. Attempted to delete profiles that didnt work.

*Screenshot(s)*
[image: profiles.png]

*A**ctual results*

Cura Crashes after i try and delete the profiles that dont work

*Expected results*
Profile work and or delete fine and or cura just works in general

*Project file*
A file does not need to be loaded it will crash regardless

*Log file*
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file
to upload, or copy a relevant snippet from it.) Atttached

*Additional information*
When my custom profile is selected in the printing area it reverts back to
draft 0.2  so i decided to try and delete my custom profiles and thats when
the app crashes everytime.
[image: image.png]

Regards

Elliot.

Ghostkeeper, 2020-05-06:
What do you mean with "delete profiles that don't work"? Did you delete profiles from the Cura interface because they don't print well?

etb1989, 2020-05-06:
Hello,

No they won't load..

Say I have a project and want to use my profile  that i have used in all
other versions of cura fine.. I go to select it from the list and it won't
load it up just reverts back to "draft 0.2".

So I figured that profile must not be compatible. So I went into setting
manage profiles and I attempt to delete that profile. And the whole app
crashes.

Regards Elliot

Elliot Bollard
0428308921
elliotbollard@gmail.com

Ghostkeeper, 2020-05-06:
E-mail attachments, like your log and those images, don't get posted on Github so we can't see them. Could you please visit your issue on [Github](https://github.com/Ultimaker/Cura/issues/7535) and post them there?

I tried your reproduction steps as follows:
* Start Cura 4.5 from a clean installation.
* Add the Tevo Tornado from the welcome screen.
* Change infill density to 42% and save that as a profile.
* Close Cura 4.5 and start Cura 4.6. The custom profile is loaded at start-up.
* Go into the profile manager and activate the Normal profile.
* Delete the custom profile. The profile is now gone. No crash though.

Ghostkeeper, 2020-05-06:
Right, yes if you start deleting files from Cura's internal configuration folder without seeing if they are still used by a machine instance, Cura will crash. This is expected behaviour.

etb1989, 2020-05-06:
Hello,

So in settings menu deleting a profile and the entire app crashing is an
expected behaviour.???

Elliot Bollard
0428308921
elliotbollard@gmail.com

Ghostkeeper, 2020-05-06:
No. If you only use Cura's interface it should never crash. But if you mess with the files in that profile folder, all bets are off.

etb1989, 2020-05-06:
I haven't messed with any config files.. It crashes through the curs
interface.

Elliot Bollard
0428308921
elliotbollard@gmail.com

etb1989, 2020-05-06:
Your reproduction is not accurate..  I don't have 4.6..

But I will download and see if it fixes my issue..

Currently in 4.5 I cannot select any other profiles other than the inbuilt
auto profiles.

0428308921
elliotbollard@gmail.com

Ghostkeeper, 2020-05-07:
> I haven't messed with any config files.. It crashes through the curs
interface.

All right, sorry then. I'll reopen this.

What are your exact reproduction steps then? For example, how did you create these profiles that were deleted? I tried creating a profile from current changes in 4.5 and deleting that profile in 4.6 but that worked fine for me.

Ghostkeeper, 2020-05-07:
I'll try again with 4.4 to 4.5.

etb1989, 2020-05-07:
Ghostkeeper,

So here is an an update as to what i have tried.

I downloaded 4.6.1

As i dont care about my custom profiles anymore i uninstalled 4.5 and when it prompted "do i want to remove configuration files" i said YES as i am wanting to start clean fresh install with no previous profiles as i have had enough so i thought a clean 4.6 would be good.

i then installed 4.6.1 all went well.

I opened up the new 4.6.1 and my custom profiles are still there..**Is this Normal???**
[image]

So again i attempt to select one of my custom profiles to use it on a print and it wont select it just reverts back to the default draft Print profile.

So i select manage Profiles (ctrl + J) and attempt to remove the all the custom profiles as at this stage they are no use to me and i just want a clean install so i can make new profiles.
when i select them the following error pops up
[image]

Then i hit send report and the cura shuts down completely.

So going forward I am committed to losing my custom profiles .
Can you please tell me a way that i can get rid of them out of version 4.6.1 so i can start fresh please.

Regards Elliot

Ghostkeeper, 2020-05-11:
To revert to factory defaults you can delete the folders `C:\Users\<username>\AppData\Roaming\cura` and `C:\Users\<username>\AppData\Local\cura`. Close Cura first though before you do that.

This is what I'd expect the uninstaller to do as well. I can't test that right now because Linux has no uninstaller, but I'll ask my colleagues if they can help out.

konskarm, 2020-05-12:
I tried reproducing the issue as follows:

Scenario 1:
- Open Cura 4.4.1
- Change printing temperature save custom profile
- Quit Cura
- Install Cura 4.5
- When asked, remove all older Cura versions
- Open Cura 4.5
- Profiles are no longer there

Scenario 2:
- Open Cura 4.5
- Change printing temperature and save custom profile
- Quit Cura
- Install Cura 4.6.1
- **Cura 4.6.1 installer does NOT ask to remove Cura 4.5**
- Open Cura 4.6.1
- 4.5 profiles are there (as expected, since 4.5 was not uninstalled)
- Profiles can be removed without a crash

So I see an issue where Cura 4.6.1 does not prompt me to uninstall Cura 4.5. Nevertheless, I cannot reproduce your issue @etb1989, since I can remove my custom profiles without a problem.

If you do not care about your custom profiles, try deleting all the configuration folders, as @Ghostkeeper suggested, and then go to control panel and uninstall all previous Cura versions (if any). Once done, try installing 4.6.1 again.

Ghostkeeper, 2020-05-14:
I see! We were missing the new versions in the uninstaller. I've changed that here: https://github.com/Ultimaker/cura-build/commit/2560b871b539fd49baad4d7d8b2ca8a25630d0a4 . This commit is in the 4.6 branch there as well.