Ultimaker/Cura #7538: Previewing box around printing area

ThatIsAPseudo, 2020-04-20:
**Is your feature request related to a problem? Please describe.**
Before launching the print, I would like to know exactly where the nozzle is going to print. It could be practical for instance in order to place tape on the bed only where there will be printed material.

**Describe the solution you'd like**
There should be a button/extension to make the nozzle describe the box that circumscribes the print, or maybe just a square on the bed.

**Describe alternatives you've considered**
I'va already tried to code a Python post-processing algorithm to parse/insert gcode at the beginning of the `.gcode` file I want to print, however I still have to edit/upload the final file manually, and my code faces some issues I can't cope with.
However, I think a "Simulate" button would be way better.

**Affected users and/or printers**
Any printer could benefit from such a feature.

**Additional context**
Here is some screenshots for the "solution" I prototyped : 
[image]

Ghostkeeper, 2020-04-22:
The layer view shows exactly where it's about to print. If your build plate has any recognisable features you can see approximately how far it's going to get. I know that this is something a lot of people like about the Ender printers and how they're displayed in Cura.

If your printer doesn't have recognisable features, there is always the grid. It's in millimetres so you could measure it out. A bit more of a hassle, but it's about the best you can do short of mounting a projector on your printer.

I believe your solution would be to print a sort of dry-run before your print, then?

ThatIsAPseudo, 2020-04-23:
Yes, for now I use the grid, but I often stop a print just after the first layer to re-center it.

> I believe your solution would be to print a sort of dry-run before your print, then?

That's exactly what I'm thinking of. To achieve this in my script, I filtered the gcode to keep only the print command lines, and I compute the min/max coordinates of each axis. 

Unfortunately, the gcode is not so easy to modify with a post-processing script. All the laser/CNC cutting machines I've used have a "dry-run" preview option, so why not a 3D printer?

Ghostkeeper, 2020-04-28:
You can do a dry run by simply setting the Flow to 0%. There is no way to then automatically continue with the actual print though after the dry run is completed aside from just queing the prints if your printer supports that.

But then again, CNC milling machines also don't automatically continue after a dry run. That would be hella dangerous!