Ultimaker/Cura #7541: Cubic Subdivision pattern not relative to model but relative to build plate

evilC, 2020-04-20:
**Application version**
4.5.0

**Platform**
Windows 10 x64 nVidia Geforce 1080 Ti

**Printer**
Ender 3

**Reproduction steps**
1. Add model to build plate
2. Ensure Cubic subdivision is set as infill method
3. Translate the model position on the build plate

**Screenshot(s)**
Same model, same settings, I just drag the model slightly left on the build plate in the 2nd picture
[image]
[image]

**Actual results**
The positioning of the infill changes

**Expected results**
No matter where you position the model on the build plate, you should get identical infill

**Additional information**
With eg Triangles infill, there are X Offset and Y Offset options, as well as a rotation option.
It would be nice to have these, but at the very least the infill pattern should be relative to the model and centered.
Each instance of the same model should have the same infill
The infill pattern should be centered on the center line of the model, such that a symmetrical model should have the same infill on each side of symmetry

Ghostkeeper, 2020-04-22:
The fact that some of these settings are not implemented for Cubic Subdivision is sort of deliberate. Cubic Subdivision uses a bit of a different routine than the rest, so we'd have to re-implement them, and we didn't yet.

I don't see much harm in this. It shouldn't be too hard to fix. I'll put it up with my colleagues to see what they think.

evilC, 2020-04-23:
Thanks, it would be really appreciated
It seems that Gyroid suffers from the same issue too, which is a real shame as these two seem to be the best strength/weight ratio for loadings in all directions, which is what I need for drone canopies in TPU which are designed to work like an airbag in a crash - at the moment it's a right pain to align the infill such that each side of the symmetrical canopy has equal strength and deformation properties
Thus, being able to rotate the infill such that there are lines that are horizontal (At the moment, the whole thing seems to be tilted at like 2 degrees) would go some way to resolving the issue, then I could just translate to line it up - a pain, but not as much of a pain

smartavionics, 2020-04-23:
> It seems that Gyroid suffers from the same issue too

My cura builds do support x/y offset for gyroid infill. You can find them at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file there for more info.

Sophist-UK, 2020-04-24:
@smartavionics - Are you willing to submit your changes as a PR?

smartavionics, 2020-04-24:
No.

mahtDFR, 2020-04-24:
Internal ticket: CURA-7394

evilC, 2020-04-26:
I see [a commit](https://github.com/Ultimaker/CuraEngine/pull/1243/commits/5d72c77d83d97de617caaae270b0c197b6ae9a14) which seems to be altering the infill origin to be relative to the middle of the mesh :)
Is there anywhere I can download this build so I can test it, or would I need to build myself? Am on windows

nallath, 2020-04-28:
You would need to build it yourself for the moment. I just merged the commit into master, so it should be available in the next release (4.7)

closed 2020-04-28 by nallath

evilC, 2020-04-28:
Thanks, did the changes go in to also allow us to rotate the infill? Without being able to align the infill such that some of the lines are perfectly horizontal or vertical, this will still result in non-symmetric infill

Ghostkeeper, 2020-04-28:
No, for now it only applies to the horizontal position, not the rotation.

Rotation would be a bit more involved here since we would need to adjust the space in which the cubes are being generated before subdividing. It won't be sufficient to pre-rotate the model and rotate it back afterwards like we do for other patterns, since we generate only one subdivided cube for the entire mesh and that subdivided cube might be too small or too big if it's rotated.

evilC, 2020-04-28:
Would it not be possible to set it such  that the pattern defaults to being aligned orthogonally? I don't really see what advantage there is to having like a 1-2deg tilt to the pattern by default

evilC, 2020-04-28:
To clarify what I am saying:
Here is a subdivided cube mesh where the faces of the cube are perfectly aligned parallel to the horizontal and vertical axes:
[image]
As you can see, the horizontal lines of the infill are not quite horizontal

Ghostkeeper, 2020-04-29:
I see what you mean, yes.

This logic is implemented here:

https://github.com/Ultimaker/CuraEngine/blob/69f0ca961b44a258c4e9e9ebf628ada044c75433/src/infill/SubDivCube.cpp#L63-L82

Perhaps there is a mathematical mistake there somewhere? It's kind of hard to understand why those constants were chosen.

evilC, 2020-04-29:
Well maths is not my strong suit by any means, but looking at the code there is a lot of stuff relating to thirds, so I guess it's maybe something to do with the number of decimal places it's calculated to?
ie .3 vs .333333333333333333
Also, do you want me to open a new issue for this?

Ghostkeeper, 2020-04-30:
I don't think the limited precision is really an issue. Those numbers have more than 50 decimal places. Even with a model of 1m you'd be talking about differences of one 1.000.000.000.000.000th of a Planck length (that is, 10^-50m). Suffice to say you wouldn't be able to see it.

But yes we could create a new issue about it.

evilC, 2020-05-01:
FYI, similar issues with support structures - see #7643