Ultimaker/Cura #7542: Cura 4.5.0 - slicing removes part of model

CaptainDario, 2020-04-20:
**Application version**
Cura 4.5.0

**Platform**
Windows 10 Pro 
AMD Rx 560

**Printer**
Creality CR 10S Pro

**Reproduction steps**
1. open the "toolholder.stl"
2. rotate it 45 degrees so that it looks like on the screenshot

**Screenshot(s)**
[image]
[image]

[CCR10SPRO_CR_10S_pro_tool_holder_v2.zip](https://github.com/Ultimaker/Cura/files/4505592/CCR10SPRO_CR_10S_pro_tool_holder_v2.zip)

**Actual results**
The backside of the toolholder gets removed.

**Expected results**
The backside is still there.

**Project file**
projet is attached as a zip as well as the .stl file
[CR_10S_pro_tool_holder_v2.zip](https://github.com/Ultimaker/Cura/files/4505597/CR_10S_pro_tool_holder_v2.zip)
[CCR10SPRO_CR_10S_pro_tool_holder_v2.zip](https://github.com/Ultimaker/Cura/files/4505597/CCR10SPRO_CR_10S_pro_tool_holder_v2.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4505577/cura.log)

**Additional information**
 The issue is a little hard to put in words but the images from Blender to Cura show the issue quite nicely I guess.

smartavionics, 2020-04-20:
The model has problems...

[image]

CaptainDario, 2020-04-21:
Okay, than I am sorry for opening this.

Why do I not get this "Model Errors" notifications?
Is there a place where I can enable this

smartavionics, 2020-04-21:
The model errors message is by a new cura feature that isn't yet available in a release.

The mesh tools message which really tells you the same thing is available in the mesh tools plugin from the marketplace.

CaptainDario, 2020-04-21:
Thank you very much for your help and time.

closed 2020-04-21 by CaptainDario