Ultimaker/Cura #7544: Cura crashes entire system

Decimator714, 2020-04-20:
I realize a previous thread related to Ryzen CPUs matches my problem pretty much exactly, but since I have intel I'm not sure what the issue is.

**Application version**
Cura 4.5

**Platform**
CPU: i7-5820k
GPU: GTX 970
Motherboard: Asus X-99 Deluxe (I'm including this because a similar report thread involving Ryzen CPUs was related to asus motherboards)

**Printer**
Creality CR-10

**Reproduction steps**
Leave the computer open with Cura, even while idling at some point the program will cause a bluescreen.

**Screenshot(s)**

**Actual results**
Bluescreen, I don't have the error message but I can get it to crash again if necessary.

**Expected results**
No crashing

**Project file**
N/A

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4506822/cura.log)
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/4506824/DxDiag.txt)

**Additional information**
The crash happened about 10 minutes after the last entry in the cura.log file.
This issue has been persistant ever since I've tried any version of cura above 4.0, I've just been using Cura 3.6 since it doesn't crash my PC. I have a 5th gen intel processor, which to my understanding is relatively rare. Maybe this could be relevant?

Ghostkeeper, 2020-04-22:
Blue screens of death are almost always caused by faulty drivers. The Intel GPU drivers are particularly infamous, but your chip doesn't have an embedded GPU.

Looking through your DxDiag there repeatedly names error code 9F as the cause of the blue screens. Error code 9F refers to [driver power state failures](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x9f--driver-power-state-failure), meaning that the driver requested their device to be set to a power state that the motherboard doesn't understand (if I understood it correctly). That sounds very similar to the plague that the Ryzen CPUs are seeing except for them it's the CPU and so the entire system crashes and it's not even able to display a BSOD. I think the "blame" is again to be placed with that motherboard not supporting C-states. You can try disabling C-states in your BIOS?

In any case, with that error code I firmly believe that the problem is not within Cura's domain, so I'll have to close this issue. I hope you can solve your firmware problems... :disappointed:

Ghostkeeper, 2020-04-22:
It also lists device driver "3" as being the problem, but there is no documentation anywhere as to which one that is.

closed 2020-04-22 by Ghostkeeper