Ultimaker/Cura #7549: no se abre el cura 2.3.1 , me da error continuamente. lo he desinstalado y vuelto a instalar. He quitao el antivirus y todo por si fuera incompatible, me iba bien y ya no.

piquerasismael, 2020-04-21:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-04-21:
Cura 2.3.1 is no longer supported. 

Next time, please fill in the issue template and use english. We won't be able to help you otherwise.

piquerasismael, 2020-04-21:
[image]
[image]

piquerasismael, 2020-04-21:
Si que es compatible con un computer de 32 bits

fieldOfView, 2020-04-21:
Unfortunately, your screenshot does not show the entire crashlog. Fortunately, I think I know the error that occurred.

Remove the file `C:\Users\[yourname]\AppData\Local\cura\cura.log` (fill in your username where it says [yourname]).

Ghostkeeper, 2020-04-22:
Het zou wel toeval zijn als dat het probleem is, maar dikke kans inderdaad.

Omdat de template niet is ingevuld zal ik deze pagina sluiten. Hopelijk lost het verwijderen van het logbestand zijn probleem op.

closed 2020-04-22 by Ghostkeeper