Ultimaker/Cura #7550: Ender3 Pro Profile Nozzle missing

Strahler85, 2020-04-21:
**Application version**
4.6

**Platform**
Windows 10 64Bit

**Printer**
Ender3 Pro

**Reproduction steps**
Create new printer Ender3 Pro

**Screenshot(s)**
[image]

**Actual results**
No Nozzle Size

**Expected results**
Nozzle Size selectable

jerryfudd, 2020-04-21:
+watching

spacecraft1013, 2020-04-21:
I have the same issue, same printer

andyscriv, 2020-04-22:
Same issue as well, no nozzle size selection :(

jerryfudd, 2020-04-22:
I don't believe the Pro profile is going to be any different to that of the Ender 3

benjareuland200, 2020-04-22:
Same issue too

Strahler85, 2020-04-22:
Found the issue, in \resources\variants there are no ender3pro files

If you copy these files in this folder than it should work
[ender3pro_variants.zip](https://github.com/Ultimaker/Cura/files/4516778/ender3pro_variants.zip)

andyscriv, 2020-04-22:
> Found the issue, in \resources\variants there are no ender3pro files
> 
> If you copy these files in this folder than it should work
> [ender3pro_variants.zip](https://github.com/Ultimaker/Cura/files/4516778/ender3pro_variants.zip)

Awsome sauce! 
Thanks for that :)

Ghostkeeper, 2020-04-22:
Thanks for the solution. That was indeed wrong. Normally we check that sort of thing before merging. It must have been an oversight (I think I was the one that merged it in this case actually...)

I've tested and added your profiles for the next release. They contain no setting overrides, but it's better than nothing I guess and the quality profiles for the Ender printers are sufficiently well designed to give decent defaults nonetheless.

See https://github.com/Ultimaker/Cura/commit/83adb98a5f3c9a51ec71bf3a8bdd41c27273fbc5.

closed 2020-04-22 by Ghostkeeper

uyesercan, 2020-04-23:
> Found the issue, in \resources\variants there are no ender3pro files
> 
> If you copy these files in this folder than it should work
> [ender3pro_variants.zip](https://github.com/Ultimaker/Cura/files/4516778/ender3pro_variants.zip)

Working very well. Thanks for solution.

Ghostkeeper, 2020-04-28:
We're probably creating a 4.6.1 release to fix a bug with the Z seam placement. We've decided to merge this fix into that as well, since it's relatively low-risk and seems to be high value based on issue reports.

hpandelo, 2020-05-09:
I can't see this menu for Anet A8 Plus.. Expected??

Ghostkeeper, 2020-05-12:
Yes, the Anet A8 Plus doesn't have any nozzle profiles to choose from. You need to adjust the settings yourself if you have a different nozzle than the standard one.

maludwig, 2020-09-17:
To those who are on a version later than 4.6 and are still experiencing this issue, I bet you're expecting that under:
Settings -> Manage Printers -> Machine Settings -> Extruder, you'll have a nozzle size.

This is not where it is. You're in the wrong spot. I feel stupid cuz this whole time I've been looking there, meanwhile, dead center of the screen it's been clearly displaying "Generic PLA - 0.4mm Nozzle", not under any setting or anything. It's just right there, just to the left of all your other print settings.

I feel dumb.

commit 83adb98a (2020-04-22): Add nozzle profiles for Ender 3 pro

These profiles were posted as solution for #7550 by Strahler85. I'm committing this in his name.

Fixes #7550.
commit a6f60df2 (2020-04-24): Add nozzle profiles for Ender 3 pro

These profiles were posted as solution for #7550 by Strahler85. I'm committing this in his name.

Fixes #7550.