Ultimaker/Cura #7555: Problem with setting of nozzle size (Anet A8)

cendakrejce, 2020-04-21:
**Application version**
4.6

**Platform**
Windows 10 

**Printer**
Anet A8

**Reproduction steps**
1. Try change nozzle size
2. Try make new FFD printer with different nozzle size

**Screenshot(s)**

[image]

**Actual results**
If I try change nozzle size it will have no effect  - start size nozzle 0.4 - change 0.6 - close window - open setting window again nozzle is 0.4

**Expected results**
Different size of nozzle

fsaitta999, 2020-04-22:
I have the same problem using the profile for Creality ender 3 pro. there is no profile for the nozzle diameter

Ghostkeeper, 2020-04-22:
Duplicate of https://github.com/Ultimaker/Cura/issues/6546.

The missing profiles for the Ender 3 Pro have been added for the next release.

fsaitta999, 2020-04-23:
> Duplicate of #6546. Try using a comma instead of a period.
> 
> The missing profiles for the Ender 3 Pro have been added for the next release.

[image]

fsaitta999, 2020-04-23:
not work

fieldOfView, 2020-04-23:
@fsaitta999 you have a different problem than @cendakrejce 
Yours is a duplicate of https://github.com/Ultimaker/Cura/issues/7550

closed 2020-04-28 by Ghostkeeper