Ultimaker/Cura #7556: [4.5 & 4.6] CURA resets each time it is started

CoinNester, 2020-04-22:
**Application version**
4.5 & 4.6

**Platform**
Windows 10

**Printer**
Rostock Max

**Reproduction steps**
1. After a hard restart of my PC, config files must've gotten corrupted somehow.
2. Restart Cura 4.5 and it starts as if for the first time even though most of the data in AppData folder remains.
3. Install 4.6 and start from scratch in setting up printer etc.
4. Restart 4.6 to check whether it saves the config, it does not. It starts as if for the first time every time. However the correct information gets stored in AppData folder
5. Uninstall every version of Cura I have and reinstall only 4.6 and the same problem persists.
6. Do the same thing with 4.5 since it worked before and it does not work correctly anymore either.

**Actual results**
Cura resets completely when restarted and does not load any profiles.

**Expected results**
Cura should start with all the printers loaded as well as print profiles

nallath, 2020-04-22:
Could you please provide the logs?

CoinNester, 2020-04-22:
[cura.log](https://github.com/Ultimaker/Cura/files/4515463/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/4515464/stderr.log)
[stdout.log](https://github.com/Ultimaker/Cura/files/4515465/stdout.log)
 Here they are

nallath, 2020-04-22:
Hmm. I do see that it's going wrong in the logs, but not what exactly. Could you try and delete your preferences file (cura.cfg, located in the AppData folder)

CoinNester, 2020-04-22:
Will do when I return from work. Thank you for the help

CoinNester, 2020-04-22:
I have removed the AppData cura folder completely and allowed Cura 4.6 to reset itself. It still has not solved my problem. I think I have figured out the bug though. Any time that I setup a new printer and I name it, the issue occurs. When I leave the original name of the printer that Cura chooses, then there is no issue and it gets saved and when I restart, everything seems to load correctly. I don't get it though, this was not the case before. 

I also tried setting up an initial printer with the name that Cura sets up and then add another printer later with the name of my choosing and the problem persists. 

Until this gets resolved I guess I'll be using "stock" names.

[cura.log](https://github.com/Ultimaker/Cura/files/4519378/cura.log)

Ghostkeeper, 2020-04-28:
I'm unable to reproduce that though. I've added a SeeMeCNC Rostock Max V3.2 printer, called it "custom name" and it was added to Cura. After rebooting, it's still there. And everything still seems to be working fine.

CoinNester, 2020-04-28:
Trust me, I don't understand it either. It only started after a hard reset I had to do that somehow erased some of my profiles at the same time and rather than corrupting the file, it was just empty inside which made no sense to me. But perhaps that's how Cura loads and unloads profiles once it is closed.

Ghostkeeper, 2020-04-28:
An empty file *is* a corrupted file if you ask me. Cura doesn't empty a file while it's loaded. It just copies it into memory.

CoinNester, 2020-04-28:
Ok understood. It is just very strange to me that only a few random profiles got corrupted and not all and not in any particular order of usage.

closed 2020-06-19 by no-response[bot]