Ultimaker/Cura #7557: (4.6.0) Z Seam Alignment and Corner Preference don't seem to be working correctly

googlematt, 2020-04-22:
**Description**
Either Z Seam alignment or Corner Preference does not seem to behave correctly.  With these settings on Sharpest Corner and Hide Seam, the seam ends up in the middle of a wall.

**Application version**
4.6.0 (worked correctly in 4.5.0)

**Platform**
Win 10, 5700 XT GPU

**Printer**
Custom Printer (Tevo Tornado)

**Reproduction steps**
1.  Open Cura 4.5.0 and 4.6.0 (perform all subsequent steps in both)
2.  Import a Benchy
3.  Choose the Draft Profile
4.  Expose the Z Seam alignment and Seam Corner Preference settings
5.  Ensure they are on "Sharpest Corner" and "Hide Seam", respectively
6.  Slice
7.  Move down to a layer in the middle of the boat hull (I used 108 for my examples)
8.  Pick a wall on the boat hull and note where the seam starts in both versions

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]
[image]

**Actual results**
Seam in 4.5 starts on a corner.  Seam in 4.6 starts in the middle of a wall.

**Expected results**
Both behave like 4.5.

**Project file**
[TT_3DBenchy_ProjectFiles.zip](https://github.com/Ultimaker/Cura/files/4513662/TT_3DBenchy_ProjectFiles.zip)

**Log file**
[CuraLogs.zip](https://github.com/Ultimaker/Cura/files/4513672/CuraLogs.zip)

**Additional information**
(Extra information relevant to the issue.)

googlematt, 2020-04-22:
I'm glad I started with a Benchy.  Would have sucked to discover this on a bigger print

smartavionics, 2020-04-22:
I think the different behaviour could be due to the change in how the polygon simplification is done. However, IMHO, using sharpest corner is pretty much a dead loss anyway, it very rarely does what you want. Far better, IMHO, is to use a user-specified z-seam hint.

googlematt, 2020-04-22:
> I think the different behaviour could be due to the change in how the polygon simplification is done. However, IMHO, using sharpest corner is pretty much a dead loss anyway, it very rarely does what you want. Far better, IMHO, is to use a user-specified z-seam hint.

I've never actually used the User Specified setting.  However, I've found that sharpest corner has been rather successful so far.

Either way, it should still work as intended, which it isn't.

googlematt, 2020-04-22:
This is the difference it makes for me.  Granted I'm still dialing things in (new printer, new filament), but this is the difference between 4.5 and 4.6.  This is with my tweaked profile, but it's the same profile across both.

[image]
[image]

smartavionics, 2020-04-22:
The problem with sharpest corner has always been that the sharpest corner can occur anywhere so when you have a wall like this...
````
------------------------------------------------------
                                                     |
                                                      -------------------------------------------------
````

(Not sure if my AA looks right for you, I think it depends on the font used)

The sharpest corner can occur at that little line segment.

I haven't actually tested this hypothesis but I think that in 4.5 and earlier the polygon simplification would have removed that little line segment but in the 4.6 version (due to https://github.com/Ultimaker/CuraEngine/pull/1238/commits/ad1dbf395c98ef1826c47826305854cbe938615d) that line segment won't get removed. Arguably, that's a regression.

googlematt, 2020-04-22:
Maybe I should have put "Regression" in my title haha.  Been a while since I worked Software QA

Ghostkeeper, 2020-04-22:
In current master the behaviour is yet again different from what you're seeing (we've made more changes to the Simplify algorithm since then), but indeed I can see the bug happening there too. It's indeed caused by a tiny hook in the contour that causes the sharpest corner to be found there.

Ghostkeeper, 2020-04-23:
I think I had 4.6 checked out by accident. This bug is fixed by https://github.com/Ultimaker/CuraEngine/pull/1214/commits/1afa4507c180caea34dd0e08d94546749842760c.

We're looking into what we can do for 4.6 now with this bug.

Sophist-UK, 2020-04-24:
It seems to me that the obvious thing is to disregard sharpest corners where one or both walls are shorter than xmm.

rburema, 2020-04-24:
We're taking this very seriously. Look for an update on this around mid-next week.

googlematt, 2020-04-24:
> We're taking this very seriously. Look for an update on this around mid-next week.

Thanks, I look forward to it!  Glad I switched to 4.6 in the middle of tuning a new printer and had clear before and after results.

nallath, 2020-04-28:
This will be fixed in 4.6.1

closed 2020-04-28 by nallath

googlematt, 2020-04-29:
@nallath @rburema thanks for the quick turnaround on this :)

brunoosti, 2020-12-02:
This problem still there in 4.8

Ghostkeeper, 2020-12-03:
Do you have a project file we can reproduce that with, Brunoosti?

brunoosti, 2020-12-03:
Take a look. I also had z seams on some layers ate the dome of a half helmet when the majority where where they should be (inside of the cut of the helmet). I've been using "user specified" because that gives me more predictability where the seam will be even though it doesn't hide it (unfortunately).
[image]

brunoosti, 2020-12-07:
Sorry! I thought only the pics would be sufficient. There you go.
You'll see that at the first few layers it lays the Z seam on the curved spot. Afterwards it changes to an inside corner for most of the model. 
I have also experienced some cases with random Z seam placement (like in that helmet case) for a couple of layers throughout the model very far from any corner where in onther "same path" layers it started on a corner as it should. Could it be for efficiency of some travels/paths?

[fanbox+duct v6 or v5 hotend.zip](https://github.com/Ultimaker/Cura/files/5652318/fanbox%2Bduct.v6.or.v5.hotend.zip)

Ghostkeeper, 2020-12-08:
You mean this bit, right? The one on the outside of the shape here?
[image]

Up until layer 32 it places the seam on this curve since those corners are the most inward corners on the model. All corners on the shape are outside corners, but these are the least outside of them, so the placement is correct.

You would usually want to put the seam on a sharp outer corner instead in this case, rather than an obtuse outer corner. This is why we added the options "Hide or Expose Seam" (sharpest corner - inside or outside) and "Smart Hiding" (sharpest corner - preferably inside but outside if necessary), both of which put the seam on the 90 degree corners on the far side of the grille.

brunoosti, 2020-12-08:
I understand what you mean. The 90 degrees angles are "more outside" than this curve. It makes sense. 
Also I went to see the helmet model I have printed on cura 4.6 (I think) and all the seams are in the innermost corner and none on the dome as they were before.
Thank you for the explanation! : ))