Ultimaker/Cura #7558: Zoom function disabled after object rotation

rruggiero, 2020-04-22:
**Application Version**
4.6.0

**Platform**
MacOS

**Printer**
Anycubic 4Max Pro

**Reproduction steps**
1. Load object on to the build platform
2. Zoom in to desired position
3. Rotate object

**Screenshot(s)**
N/A

**Actual results**
Once you rotate the object the window will zoom out and you are unable to zoom in again.  You can only zoom out.  To reset the position select View->Camera Position->3D View and you will once again be able to zoom in and out.
 
**Expected results**
Zoom should not have  changed and I should be able to zoom in and out as required.

**Project file**
N/A

**Log file**
N/A

**Additional information**
N/A

Ghostkeeper, 2020-04-22:
I'm unable to reproduce this problem. I can zoom in, rotate the object, then zoom out again. Maybe the problem is only on MacOS? Is someone else able to reproduce the problem?

Recording here: http://dulek.net/work/zoom-rotate-unzoom.mkv

rruggiero, 2020-04-22:
After seeing your message I tried to do a screen recording of it on my Mac and it worked fine with quicktime recording the event.  I could not use the functionality at all yesterday and yet this morning I can. I am fairly sure that I rebooted yesterday and still experienced the problem but maybe I didn’t reboot.

My apologies, please close off this report.

Ghostkeeper, 2020-04-22:
Maybe something weird with the touch pad or something. I guess we'll keep an eye out in case it happens again.

closed 2020-04-22 by Ghostkeeper

rruggiero, 2020-04-24:
Hi,

The problem has returned. and I have  a video which I can share for you (attached).

I can "zoom out" but I have limited "zoom in" ability. When I added the 3rd object on to the build plate I selected from the bottom left camera positions until I could find the angle I wanted and then when I selected the third or fourth angle the problem returned and I could no longer "zoom in".

I am unable to reboot my computer at the moment as I have an unfinished job processing in the background but the machine details are as follows:

**OS**  macOS High SIerra Version 10.13.6
iMac (27-inch, Mid 2011)
**Processor**  3.4 GHz Intel Core i7
**Memory**  32 GB 1333 MHz DDR3
**Graphics**  AMD Radeon HD 6970M 1024 MB 

[Cura problem.zip](https://github.com/Ultimaker/Cura/files/4526664/Cura.problem.zip)

Cheers,

Rick

rruggiero, 2020-04-24:
Hi,

Just to ket you know the problem returned and I put a video on the GitHub page.

Cheers,

Rick Ruggiero

e:  rick.ruggiero@gmail.com <mailto:rick.ruggiero@gmail.com>

Ghostkeeper, 2020-04-28:
I think the problem here is really that the camera origin is moved very far away from the build volume. You can rotate the camera around the camera origin and zoom towards the camera origin, but you can't zoom beyond the camera origin.

A simple way to fix it is to click one of the buttons in the bottom left that reset the camera position. That's the main reason why the buttons are there, really: To help when the camera got into a weird position.

rruggiero, 2020-04-30:
Yes, that is exactly how I reset the camera position.  But it is frustrating as if you are inspecting support areas for possible editing you end up losing position and I just end up giving up and print without the support.  But when this happens I am not really zoomed out and mostly zoomed in.

Cheers,

Rick Ruggiero

e:  rick.ruggiero@gmail.com <mailto:rick.ruggiero@gmail.com>

Ghostkeeper, 2020-05-04:
The problem is really that you're already fully zoomed in. Just not zoomed into the position you meant to, but an empty spot in front of it. And the model happens to be on the background far away.

rruggiero, 2020-05-06:
Hi,

Actually I wasn’t because after I reset the screen I could zoom in much further. 

Kindest regards,

Rick Ruggiero
e: rick.ruggiero@gmail.com
m: 0450 025 425
---

Ghostkeeper, 2020-05-07:
Clicking on those buttons indeed also resets the zoom.