Ultimaker/Cura #7560: Deleting a printer profile do not clean up all files

DivingDuck, 2020-04-22:
**Application version**
v4.6 and older versions of v4.x too

**Platform**
Windows 10 64bit

**Printer**
Different printers, it doesn't matter

**Reproduction steps**
Create a new printer profile from an existing definition e.g. Creality, Prusa ...
Then activate the new, deactivate and delete this profile.
**Screenshot(s)**
N/A

**Actual results**
Cura leave definition files (at least I found them there) at:
c:\Users\USER\AppData\Roaming\cura\4.6\definition_changes\
c:\Users\USER\AppData\Roaming\cura\4.6\extruders\
c:\Users\USER\AppData\Roaming\cura\4.6\quality_changes\
c:\Users\USER\AppData\Roaming\cura\4.6\user\
c:\Users\USER\AppData\Local\cura\4.6\cache\definitions\4.6.0\

**Expected results**
After deleting a (printer) profile I expect not having orphan files lying around somewhere. At least it would be an benefit, having a maintenance routine that tells me about those files and gave me the possibility to delete them selective without the need deleting all profiles.

I got notice about these files because I was testing a new plugin from ahoeben. Please see as reference:
 https://community.ultimaker.com/topic/32126-cura-startup-optimiser-plugin-needs-testers/

**Project file**
N/A
**Log file**
N/A
**Additional information**
Cura is installed on 3 PC's and after checking each of them I found on every PC those orphan files from older versions or from testing a profile for all Cura 4.x versions.
Some of them seems to be there because I open an .3mf file from an external source or from forum users for testing . (At least a guess, because I don't have an other explanation for some of these profile files).

fieldOfView, 2020-04-22:
Minor correction: you are removing a printer instance, not a printer definition.

DivingDuck, 2020-04-22:
Yes, thanks for this correction. Sometimes it is a bit difficult for me to name/translate it the correct way. :)

rburema, 2020-04-24:
Devs: Internal ticket made as CURA-7398

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

MariMakes, 2023-07-10:
Hey @DivingDuck,

Thanks for your report 👍
I took a look at Cura 5.4 and I cannot reproduce your issue anymore. 
Here is what I did:

1. I started with a clean install
2. I added a Creality CR-10
3. I sliced something with it. 
4. I selected a different printer
5. I deleted the CR-10
6. I visited my configuration folder
7. I searched for any left over reality files

But it couldn't find any files 🤔 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

DivingDuck, 2023-07-12:
Hi @MariMakes,
thanks for coming back to me and sorry for the late answer.

I just tested it again and with 5.4.0 deleting a printer after installing it work fine now. In addition I test what happen if a printer setup was installed via a project file and this as well is working fine now.
The only missing part are older existing profiles. These are still there (quite big amount of profile files). I guess I need to do some manual clean-up as these seems to be now a lot of orphans (guess mostly from project files of former Cura releases).

The last test I wanted to do was checking this for material profiles, but unfortunately this one I can't test as Cura crashed as soon I click on manage materials. I have let Cura send an error report for this. 

So I guess most of the issue is solved now.

MariMakes, 2023-07-17:
Thanks for the update 👍 
I'll close this issue.