Ultimaker/Cura #7562: Print USB Abort does not send Machine end Gcode

Sophist-UK, 2020-04-22:
**Application version**
4.5.0

**Platform**
Windows

**Printer**
Dagome DiscoEasy200

**Reproduction steps**
1. Printed something by USB
2. Aborted the print

**Actual results**
Print head returned to standard home location rather than location specified in custom end gcode set for the printer.

**Expected results**
Cura should send the custom Gcode specified for the printer.

rburema, 2020-04-23:
Should the behaviour of aborting a print be different when you do it from Cura versus when you do it from a printer? If no, then how would the second case be handled for all printer types?

Sophist-UK, 2020-04-23:
Is there a standard set of Gcode run when you cancel the print from your printer i.e. same Gcode on every printer? I doubt it!

The code for cancelling a print is at https://github.com/Ultimaker/Cura/blob/b1e95d7a8009d4378ae54e1928051457c72738d8/plugins/USBPrinting/USBPrinterOutputDevice.py#L360 and is
```
M140 S0 ; Set Bed Temperature
M104 S0 ; Set HotEnd Temperature
M107 ; Fan off
HomeHead ; Not sure what is produced for this if anything - perhaps G0 X Y
M84 ; Disable steppers
```
This might be a good default if there is no Machine End Gcode, but IMO if there is machine End Gcode then that should be used instead.

Sophist-UK, 2020-04-23:
On reflection, if there is no Machine End Gcode, we would have the assume that the machine has its own in-built end routine, in which case the should Cura really issue any default Gcode?

Sophist-UK, 2020-04-23:
See also #7564 - it also occurs to me that if you cancel a (USB) print job, then not only should you send the Machine End Gcode, you should also leave the extruders in the same state they would be in if the print had run to completion.

Ghostkeeper, 2020-04-24:
That depends. If the end g-code contains some movements to e.g. wipe off the nozzle or something and you abort the print because there is a layer shift, you really want the aborting to just stop everything rather than do a whole bunch of finishing and cleaning. Finishing and cleaning can do damage to your printer and your prints.

Sophist-UK, 2020-04-24:
Damaging a print that has been aborted may not be an issue (unless it damages a previous object in print One-at-a-Time. Damaging the printer is (obviously) not a good idea.

Do we need Machine / Extruder Abort Gcode settings, and use the above defaults if Machine abort Gcode is not set?

ashleysommer, 2020-04-24:
I think there is a distinction between stopping a print, and finishing a print.
The Machine End gcode should be run when a print is Finished, but not necessarily when a print is stopped.
I think stopping a print is akin to an "about" process, where the finsihing is more of a "finalization" process. Perhaps it should be renamed as such.
I suppose there could even be machine-specific abort code that could be set on a per-machine basis too.

Sophist-UK, 2020-04-24:
I am struggling to see what generic end Gcode might do successfully with a completed print that would damage the machine if you had a partial print. Yes - you might end up doing unnecessary stuff, but I can't see how it would break something.

However, I think it likely that start and end Gcode are set to match - for example, if you retract filament at the end of a job (to enable a change of filament beween jobs) your start gcode will have matching instructions to reinsert it. (Extruder start / end Gcode does exactly this!!)

And for me, it is seems important that the printer should be left in a state that enables printing to start again without manual setup (other than clearing the bed).

THX2112, 2020-05-16:
Abort Print should have it's own user modifiable gcode settings just like the "start" gcode and "end" gcode.

Printing over USB is initiated by Cura the printers gcode doesn't always run when a print ends by aborting it from Cura (like mine, which is why I'm here). 

The user should be able to customize this, because not every printer will act the same way, and users may want specific things done if a job is aborted.

GregValiant, 2026-05-30:
I was going through these older reports and wondering about each.  Some were duplicates that were missed when a fix was made.

This one is difficult to imagine as "aborting" from Cura would need a separate routine to pop in and be sent and it seems to me that it would be a "personal preference" sort of thing where different users would want it done in different ways.  It would almost seen to need a popup where the user would confirm an option to "Send Ending G-Code".  That would probably need to send either a user defined script, or the Ending G-Code that is present in the active printer definition.  Somehow the script would probably need to be saved with the project.

I'll leave it open but with Octoprint and other print servers available, I don't see this getting much attention.  It seems like it would be a lot of work and not affect a lot of people.